Anime4000/RTL960x #212: Can an ODI module register with an OLT device simultaneously using multiple ONU configurations?Services are isolated by vlan.

wuchuwei, 2023-11-29:
Our company needs to buy a lot of broadband from carriers.However, due to the carrier's restrictions, only eight pppoe accounts can be registered on an ONU.We now need more ODI modules to expand our business.Since the ODI module is very hot during work, it is necessary to maintain a certain distance when installing the ODI module to the optical switch to allow the ODI to dissipate heat faster, so that the port utilization of the switch is very low.Since our business traffic is not very large and the performance of ONU does not need to be very high, we want to find a way to replace multiple ONUs with one ONU.I wonder if the ODI module can do this by changing the code?
我们公司需要从运营商上购买很多宽带。但由于运营商的限制一个ONU只能注册8个pppoe帐号.。现在需要更多的ODI模块才能扩大我们的业务。由于ODI模块在工作中发热很大，在安装ODI模块到光交换上时不得不保持一定的距离让ODI散热更快，这样做交换机的端口利用率就很太低了。由于我们的业务流量不是很大ONU性能不需要很高，所以想寻找一种用一个ONU代替多个ONU的办法。不知道ODI模块能否通过更改代码实现这个目的？

prime-noc, 2023-12-18:
No. But You need high speed fan switch to solve the heat problem.

不太行，你可以找个风扇多的交换机来专门接猫棒，就可以避免发热的问题，不要用TPLINK或MikroTik那种的交换机。
比如搞个二手Arista 7050S或Dell S4810P之类的模块散热应该就会好很多

akw28888, 2023-12-18:
No. But You can use switch that contains high speed fans to solve the heat problem.

不太行，你可以找个风扇多的交换机来专门接猫棒，就可以避免发热的问题，不要用TPLINK或MikroTik那种的交换机。
比如搞个二手Arista 7050S或Dell S4810P之类的模块散热应该就会好很多