Anime4000/RTL960x #217: Opinions on FS GPON-ONU-34-20BI / Source Photonics SPS-34-24T-HP-TDFO

Strykar, 2023-12-05:
These are Lantiq SoCs running OpenWRT apparently, I am wondering why they are not more popular?
Is it the lack of 4-port ONU emulation?

- Source Photonics SPS-34-24T-HP-TDFO - https://www.sourcephotonics.com/wp-content/uploads/2017/08/DS-8085-02_SPS-34-24T-HP-TDFO.pdf
- FS GPON-ONU-34-20BI - https://www.fs.com/products/133619.html

ToTheCLI, 2023-12-05:
Pros Lantiq SoCs
1-DDM readings like Rx, Tx, Temp are available  (ODI Unavailable) 
2-Fiber SC/APC receptacle (ODI SC/UPC needs ugly converter)
3-Is more compatible with Huawei and similar OLTs
Cons Lantiq SoCs
1-Only SSH unless Right.CN firmware is used
2-Less Configuration options and might be less Compatible with some OLTs
3-Some have noted reduced upload speed on high latency servers (Not sure if true)

For me Lantiq is the better option unless it's incompatible with the OLT

Strykar, 2023-12-06:
Thanks, do any of the FS GPON ONU / ONT's have 4-port emulation?
See - https://www.fs.com/c/gpon-3660

ToTheCLI, 2023-12-06:
> Thanks, do any of the FS GPON ONU / ONT's have 4-port emulation? See - https://www.fs.com/c/gpon-3660

No and the one below is the Only one that can be used instead of ISP's Modem
https://www.fs.com/products/133619.html

Kozino1, 2024-02-22:
> if you internet at LAN 2, Lantiq will disconnect from SFP host!

Thank you for this valuable info.  This explains why I cannot use this stick.

Strykar, 2024-02-24:
> > if you internet at LAN 2, Lantiq will disconnect from SFP host!
> 
> Thank you for this valuable info. This explains why I cannot use this stick.

Sorry to hear it.
@Anime4000's suggestions are spot on, the ODI / HSGQ SFP just worked to replace my ISP router / optics.