Anime4000/RTL960x #224: PCI-E card that compatible with ODI Stick?

itsmehoaq, 2023-12-16:
Can you suggest a compatible GPON stick and recommend a model that I should purchase (Intel 82599ES (R86S-N) or Mellanox ConnectX-3 (R86S))?

Anime4000, 2023-12-16:
this hard to answer, since 2.5 and 5Gbps is relative new, it existed after 10Gbps...

however, BCM57810S is what most people choose for 2.5Gbps

if not planing to 2.5Gbps, 1Gbps is supported by default for most SFP PCIe Card

itsmehoaq, 2023-12-16:
I thought 2.5Gbps existed before 10Gbps. After a quick Google search, I found that neither of the NICs I listed supports 2.5Gbps.

itsmehoaq, 2023-12-16:
maybe I should just buy a converter and use an RJ45, but that would take up one slot for the AC-DC adapter and some space :(

Anime4000, 2023-12-16:
> I thought 2.5Gbps existed before 10Gbps. After a quick Google search, I found that neither of the NICs I listed supports 2.5Gbps.

after 100Mbps, there 1Gbps, then 10Gbps (2002)...

2.5Gbps and 5Gbps exist in year 2015 after WiFi 5 (802.11ac) existed because WiFi 5 much faster then 1Gbps LAN cable

It use 10Gbps signaling and slow down to 50% for 5Gbps, and 25% slow down for 2.5Gbps

> maybe I should just buy a converter and use an RJ45, but that would take up one slot for the AC-DC adapter and some space :(

Yes, is nice to have converter in-case incompatible and troubleshooting

itsmehoaq, 2023-12-16:
> people hack BCM57810S NVRAM to support 2.5G by reducing 10Gbps signaling speed to 25%

Can you tell me where the NVRAM hack documentation is? I'm unable to find any related information in your repo..

Anime4000, 2023-12-16:
https://www.dslreports.com/forum/r32230041-Internet-Bypassing-the-HH3K-up-to-2-5Gbps-using-a-BCM57810S-NIC

ROS x86? I not sure, maybe not

itsmehoaq, 2023-12-17:
> https://www.dslreports.com/forum/r32230041-Internet-Bypassing-the-HH3K-up-to-2-5Gbps-using-a-BCM57810S-NIC

After reading this, I am certain that ROS x86 does not support 2.5 Gbps because it requires a kernel patch.

Anime4000, 2023-12-17:
if plant to use ROS x86, why not bought RB5009 ?

itsmehoaq, 2023-12-18:
> if plant to use ROS x86, why not bought RB5009 ?

I want do some docker stuff

Anime4000, 2023-12-18:
> > if plant to use ROS x86, why not bought RB5009 ?
> 
> I want do some docker stuff

your docker need is specifically build for x86?

I have docker run on RB5009
[image]