Anime4000/RTL960x #225: How to enable the 4-port emulation feature on HSGQ DFP-34X-2C2

zongmingye, 2023-12-17:
[image]
[image]
I have two pppoe accounts that use the same onu’s loid. I would like to know how to set up simultaneous PPPoE dialing on ROS. Can enabling the 4-port emulation to dial at the same time? currently, only one account can successfully dial at a time using the same port

Anime4000, 2023-12-17:
both are come same VLAN ? or not using VLAN?

zongmingye, 2023-12-17:
Both used same vlan id 41
[image]

Anime4000, 2023-12-18:
I think better make it transparent bridge and add VLAN at Mikrotik instead.
This what do with my internet:
[image]

zongmingye, 2023-12-19:
In the end,the problem was fixed by useing vrrp interface
[image]

closed 2023-12-19 by zongmingye

Anime4000, 2023-12-20:
wow, your ISP need VRRP to work?
we here just use same VLAN with many PPPoE Client

can you show this page?
[image]