Anime4000/RTL960x #229: ODI - Issue with SFU Firmware

tumix83, 2023-12-21:
Hello, 

The ISP "Bouygues FR" provided an Nokia G 010G Q, the router "OPNSENSE"  is connecting via DHCP using the ISP VLAN

That I'm trying to replace with an ODI stick but its only work with the IGD Firmware (hybrid mod), 
The SFU firmware mode seems to be more suitable in my configuration, i have O5, VLAN with mibs 84 / 171 but no internet....

Here all parameters i use :

```
flash set VLAN_CFG_TYPE 1
flash set VLAN_MANU_MODE 0
flash set DEVICE_TYPE 2 (with IGD Firmware)
flash set DEVICE_TYPE 0 (with SFU Firmware)

flash set ELAN_MAC_ADDR 605747E89B14
flash set MAC_KEY 3f51423520fe7d73ba20bf1dfdbd9893
flash set OMCI_OLT_MODE 21

flash set GPON_PLOAM_PASSWD 0000013XXXXXXXXXXXXX
flash set GPON_SN ALCLF9EFCC12
flash set PON_VENDOR_ID ALCL

flash set OMCI_SW_VER1 3FE49494AOCK21
flash set OMCI_SW_VER2 3FE49494AOCK21

nv setenv sw_custom_version0 3FE49494AOCK21
nv setenv sw_custom_version1 3FE49494AOCK21

flash set HW_HWVER 3FE49476ABAA01
flash set GPON_ONU_MODEL G-010G-Q
flash set OUI 605747

flash set HW_SERIAL_NO M082224ALU00072818
```

**SFU Firmware :**

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 100
FwdOp:  0x10
NumOfEntries: 1
=================================

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
[... 277 lines cut ...]
Dis        0

RTK.0> command:l2-table get entry address 980

LUT address: 0x03d4 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
A4:7B:2C:84:63:E0 2   1   7   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:l2-table get entry address 984

LUT address: 0x03d8 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:00:5E:00:01:A8 2   1   7   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

RTK.0> command:DONE! L2 Table Print Valid
```

I'm missing something?
Thank in advance !

ToTheCLI, 2023-12-23:
SFU firmware does not respect (follow) mib171 VLANs so in order to use it you have to use a different VLAN (10 or 100 probably)

tumix83, 2024-01-07:
Thank for reply, how i should specify the correct VLAN ? on OPNSENSE or ODI ?
The VLAN for internet is 100

Thank in advance

ToTheCLI, 2024-01-07:
> Thank for reply, how i should specify the correct VLAN ? on OPNSENSE or ODI ? The VLAN for internet is 100
> 
> Thank in advance

Better to set the VLAN on OPNSENSE and leave the SFP on transparent.

tumix83, 2024-01-10:
Fixed with parameter : 
`flash set OMCI_CUSTOM_BDP 0`
Now its work with SFU Firmware

Anime4000, 2024-01-12:
Mark as solved, added this into Troubleshooting Readme

Feel free to re-open is anyone have same issue

closed 2024-01-12 by Anime4000