Anime4000/RTL960x #230: dfp-34x-2c2 can't see ISP gateway

Smotrov, 2023-12-22:
Bought dfp-34x-2c2 instead of bricked V2801F (which worked for a while)
ISP white listed MAC of dfp-34x-2c2.  My ISP is EPON with static IP.

Link Is OK and I can see some packets. 

[image]
[image]

However, I just can't ping ISP gateway... 

It supposed to replace ISPs  PON 

[image]

Any ideas would be highly appreciated.

Smotrov, 2023-12-23:
Besides, it is interesting which settings are essential for EPON ISP (apart from MAC address). Everybody talking about GPON mostly...

Anime4000, 2023-12-25:
> Besides, it is interesting which settings are essential for EPON ISP (apart from MAC address). Everybody talking about GPON mostly...

Sadly yes, almost everyone use GPON including me 😅

So, documentation about EPON is very limited here, perhaps you can try and find out

if you have Discord, join this chat forum (7 days limit start from today):
https://discord.gg/WQQEUvFM

korzhyk, 2024-02-02:
@Smotrov what ISP do you use?

Smotrov, 2024-02-05:
I'm on Triolan Ukraine. Now everything works perfectly fine. The only thing was to properly set VLAN. Either on SFP module or on the Interface on the router. 

I find VLAN ID I just did "touch" on SFP interface and from there VLAN ID become obvious :-)

closed 2024-02-05 by Smotrov