Anime4000/RTL960x #231: Documentation is unclear

Smotrov, 2023-12-23:
I'm desperately trying to setup dfp-34x-2c2 from my EPON ISP (Triolan Ukraine) (on RB8009)
Few lines in the [documentation](https://github.com/Anime4000/RTL960x/blob/main/Docs/Setup_Stick.md) seams clear to me. 
According to My ISP the only thing the use to authenticate is my MAC address.

> If your internet is Bind MAC Address, then you need change Router WAN MAC Address.
What does it mean _internet is Bind MAC Address_?  What exactly is _change Router WAN MAC Address_? Change SFP cage MAC? To which value? I was thought that ISP can see only SNF Stick MAC, why should I change Router WAN MAC Address?

> EPON might need change MAC Address, use V2801F stick for EPON connection
What does it mean "use V2801F"? I can not use dfp-34x-2c2 stick? I believe it suppose to work with both EPON/GPON?

Any clarifications would be highly appreciated.

Anime4000, 2023-12-25:
> If your internet is Bind MAC Address, then you need change Router WAN MAC Address.

Some ISP locking by MAC Address either from PON Stick or SFP Interface (if using PPPoE)

> EPON might need change MAC Address, use V2801F stick for EPON connection

if your internet GPON, ignore it. EPON is just Ethernet on PON, Layer 2 transport which is carry MAC Address like any Ethernet would, 

very different from GPON which is Layer 3 transport which is carry IP Address, GPON has own layer transport which is MAC Address of ONT begin stripped off, and GPON behave like xDSL which is using ATM (Asynchronous Transfer Mode) but better. So, GPON has GEM Encapsulate which didn't carry MAC Address of ONT. Only carry MAC Address of WAN Router

Smotrov, 2023-12-26:
> authenticate you with just router MAC address, 

Hmm. You mean router SFP port MAC address? Not SFP Stick MAC address?

Smotrov, 2023-12-27:
I've spent couple of hours with my ISP but with no luck. 
The do see my registration of MAC address and white list it. 
However whatever I do I can not ping default gateway, thus there is no internet. 

My ISP ONU has following settings. 

[image]
[image]
[image]
[image]

Any help would be highly appreciated. 

Again,  V2801F in my case was working out of the box on default settings (until it bricked).

Smotrov, 2023-12-27:
This was the first I did in the Web UI. Doesn't help thought.

Smotrov, 2023-12-27:
Sure @bastietm ! 
<details>
  <summary>SFP Stick settings</summary>
  
[image]
[image]
[image]
[image]
[image]

</details>

<details>
  <summary>Router config</summary>
[admin@MikroTik] > /export
# 2023-12-28 01:30:12 by RouterOS 7.13
# software id = 82T1-PVIK
#
# model = RB5009UPr+S+
# serial number = HF5PPPPPPPPP
/interface bridge
add admin-mac=78:9A:18:70:XX:XX auto-mac=no comment=defconf name=bridge-lan port-cost-mode=short
/interface ethernet
set [ find default-name=ether1 ] comment="ISP ComfortNet" poe-lldp-enabled=yes poe-out=off
set [ find default-name=ether2 ] poe-out=off
set [ find default-name=ether7 ] comment="Router 1st floor"
set [ find default-name=ether8 ] comment="Router 2nd foor" poe-lldp-enabled=yes poe-priority=1
set [ find default-name=sfp-sfpplus1 ] comment="ISP Triolan"
/interface list
add comment=defconf name=WAN
add comment=defconf name=LAN
/interface wifi channel
add band=2ghz-ax disabled=no name=2G skip-dfs-channels=disabled width=20/40mhz
add band=5ghz-ax disabled=no name=5G skip-dfs-channels=disabled width=20/40/80mhz
/interface wifi datapath
add bridge=bridge-lan disabled=no name=Datapath
/interface wifi security
add authentication-types=wpa2-psk,wpa3-psk disabled=no ft=yes ft-over-ds=yes group-key-update=1h name=PrivateSecurity
/interface wifi configuration
add channel=5G country=Ukraine datapath=Datapath disabled=no mode=ap name=5GConfig security=PrivateSecurity ssid=\
    "AirPort Extreme" tx-power=23
add channel=2G country=Ukraine datapath=Datapath disabled=no mode=ap name=2GConfig security=PrivateSecurity ssid=\
    "AirPort Extreme Slow" tx-power=25
/interface wireless security-profiles
set [ find default=yes ] supplicant-identity=MikroTik
/ip pool
add name=default-dhcp ranges=192.168.88.10-192.168.88.254
/ip dhcp-server
add address-pool=default-dhcp interface=bridge-lan lease-time=10m name=defconf
/routing table
add disabled=no fib name=Comfort-route
add disabled=no fib name=Triolan-route
/interface bridge port
add bridge=bridge-lan comment=defconf disabled=yes interface=ether2 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether3 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether4 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether5 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether6 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether7 internal-path-cost=10 path-cost=10
add bridge=bridge-lan comment=defconf interface=ether8 internal-path-cost=10 path-cost=10
/ip neighbor discovery-settings
set discover-interface-list=LAN
/interface detect-internet
set detect-interface-list=all
/interface list member
add comment=defconf interface=bridge-lan list=LAN
add comment="ISP Triolan" interface=sfp-sfpplus1 list=WAN
add comment="ISP ComforNet" interface=ether1 list=WAN
/interface wifi capsman
set ca-certificate=auto enabled=yes package-path="" require-peer-certificate=no upgrade-policy=none
[... 99 lines cut ...]
add action=accept chain=forward comment="defconf: accept ipsec AH" protocol=ipsec-ah
add action=accept chain=forward comment="defconf: accept ipsec ESP" protocol=ipsec-esp
add action=accept chain=forward comment="defconf: accept all that matches ipsec policy" ipsec-policy=in,ipsec
add action=drop chain=forward comment="defconf: drop everything else not coming from LAN" in-interface-list=!LAN
/system clock
set time-zone-name=Europe/Kiev
/system note
set show-at-login=no
/system package update
set channel=testing
/tool bandwidth-server
set authenticate=no
/tool graphing interface
add interface=bridge-lan
add interface=*B
add interface=*C
add interface=ether1
add interface=sfp-sfpplus1
/tool mac-server
set allowed-interface-list=LAN
/tool mac-server mac-winbox
set allowed-interface-list=LAN
</details>

Meanwhile, I've tried to add VLAN ID on RB5009 side like [here](https://github.com/Anime4000/RTL960x/blob/main/Docs/更換%20SFP%20GPON%20ONU%20(Hinet)%20CCR1009-7G-1C-1S%2B%20-%20HackMD.pdf) which no luck so far...

Smotrov, 2023-12-28:
> I assume you set it up in bridge mode

Does it require any additional settings on SFP through telnet?

Anime4000, 2023-12-28:
perhaps try flash this firmware instead:
https://github.com/Anime4000/RTL960x/raw/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_hybrid_221209.tar

The WebGUI look same as V2801F

Smotrov, 2023-12-28:
> perhaps try flash this firmware

Thank you! I did. But with no luck. By the way can't see where to VLAN ID there. But this is not a problem, as soon as I can set it on RB5009 side.

Smotrov, 2023-12-28:
Taking to account that V2801F (even without touching VLAN) was working out of the box, maybe if I'll set all **flash get / set** on dfp-34x-2c2 the way it set on default V2801F it would finally work? 

Is anybody has default settings of V2801F?

Anime4000, 2023-12-29:
similar stick to V2801F is https://www.leolabs.pl/ont-leox-lxt-010s-h.html

many ODI firmware runs on GPON mode sadly

Smotrov, 2023-12-29:
Is it possible to reset dfp-34x-2c2 to all factory default? Including  config, flash, ELAN_MAC, VS_AUTH, MAC_KEY etc.

Smotrov, 2024-01-02:
Still fighting with my SFP. For various reasons I believe that the problem is in the way dfp-34x-2c2 works with VLAN.
ISP ONU works perfectly well without any problems. So I managed to extract its settings from the dump. 

Is there any way to replicate VLAN settings like it is done in ISP ONU? 

<details>
  <summary>ISP ONU settings dump</summary>

```
<DB>
    <Tbl name="DHCPCComm" RowCount="1">
        <Row No="0">
            <DM name="BootpFlag" val="0" />
            <DM name="Dscp" val="-1" />
            <DM name="Priority" val="-1" />
            <DM name="QueueNum" val="15" />
            <DM name="SendRelease" val="0" />
        </Row>
    </Tbl>
    <Tbl name="LAND" RowCount="4">
        <Row No="0">
            <DM name="ViewName" val="IGD.LD1" />
            <DM name="LANDName" val="LAN" />
            <DM name="Name" val="br0" />
            <DM name="Enable" val="1" />
            <DM name="IPAddr" val="192.168.1.1" />
            <DM name="NetMask" val="255.255.255.0" />
            <DM name="IPv6Addr" val="fe80::1" />
            <DM name="IPv6PrefLen" val="64" />
            <DM name="IPv6GUAAddr" val="::" />
            <DM name="IPv6GUAPrefLen" val="128" />
            <DM name="AddressingType" val="1" />
            <DM name="HelloTime" val="10" />
            <DM name="ForwardDelay" val="15" />
            <DM name="MAXAge" val="20" />
            <DM name="AgeingTime" val="300" />
            <DM name="RootPathCost" val="0" />
            <DM name="MAC" val="c0:7e:40:b6:5c:65" />
            <DM name="STPEnable" val="0" />
            <DM name="IGMPSNPEnable" val="1" />
            <DM name="IfMtoU" val="1" />
            <DM name="IfQrySend" val="1" />
            <DM name="IfSendWan" val="0" />
            <DM name="UnregisMode" val="0" />
            <DM name="GrpTimeout" val="360" />
            <DM name="LeaveMode" val="1" />
            <DM name="IgmpQueryVersion" val="2" />
        </Row>
        <Row No="1">
            <DM name="ViewName" val="" />
            <DM name="LANDName" val="" />
            <DM name="Name" val="" />
            <DM name="Enable" val="1" />
            <DM name="IPAddr" val="0.0.0.0" />
            <DM name="NetMask" val="0.0.0.0" />
            <DM name="IPv6Addr" val="fe80::1" />
            <DM name="IPv6PrefLen" val="64" />
            <DM name="IPv6GUAAddr" val="::" />
            <DM name="IPv6GUAPrefLen" val="128" />
            <DM name="AddressingType" val="1" />
            <DM name="HelloTime" val="10" />
            <DM name="ForwardDelay" val="15" />
            <DM name="MAXAge" val="20" />
            <DM name="AgeingTime" val="300" />
            <DM name="RootPathCost" val="0" />
            <DM name="MAC" val="00:00:00:00:00:00" />
            <DM name="STPEnable" val="1" />
            <DM name="IGMPSNPEnable" val="1" />
            <DM name="IfMtoU" val="1" />
            <DM name="IfQrySend" val="1" />
[... 1412 lines cut ...]
            <DM name="MaxSrcPort" val="0" />
            <DM name="DstIp" val="0.0.0.0" />
            <DM name="DstIpMask" val="0.0.0.0" />
            <DM name="MinDstPort" val="0" />
            <DM name="MaxDstPort" val="0" />
            <DM name="FilterTarget" val="0" />
            <DM name="InOut" val="1" />
            <DM name="LocalForward" val="2" />
            <DM name="Level" val="3" />
            <DM name="Order" val="6" />
        </Row>
    </Tbl>
    <Tbl name="OAMdebug" RowCount="1">
        <Row No="0">
            <DM name="OAMLogCTRL" val="0" />
            <DM name="EXOAMLogCTRL" val="0" />
        </Row>
    </Tbl>
</DB>

```

</details>

Will infinitely appreciate any ideas.

Anime4000, 2024-01-03:
have you tried to run `omcicli mib get` `84` and `131` just in case

Smotrov, 2024-01-03:
Yep. I did. It gives empty response 
```
BusyBox v1.12.4 (2022-09-23 19:33:44 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
# omcicli mib get 131
# 
```

Smotrov, 2024-01-03:
Here is other output. Ac I can understand no VLAN at all...

<details>
  <summary>l2-table get entry address valid</summary>

```

RTK.0> l2-table get entry address valid 

LUT address: 0x0008 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
78:54:2E:07:64:63 2   0   6   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x004c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
F4:8C:EB:EB:67:0C 2   0   2   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0050 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:24:01:30:CD:CC 2   0   3   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0068 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash

--More--
08:55:31:77:EA:EB 2   0   6   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0084 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
1C:BD:B9:D8:59:B8 2   0   4   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x00a8 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
60:63:4C:4F:1A:30 2   0   7   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x00ac (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:AD:24:0D:F9:56 2   0   6   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp

--More--
Dis    Dis  Dis     Dis     Dis

LUT address: 0x00d4 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
1C:BD:B9:D7:14:EE 2   0   4   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x00dc (1K LUT)
[... 281 lines cut ...]
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
38:3A:21:28:A9:56 3   0   7   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x03d1 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
4C:5E:0C:C8:E6:C4 2   0   7   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp

--More--
Dis    Dis  Dis     Dis     Dis

LUT address: 0x03d8 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
1C:BD:B9:D6:F6:CE 2   0   6   1    Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

```

</details>

Smotrov, 2024-01-03:
When set VLAN mode to automatic on the module Web-UI I can see VLANs on the interface. 

[image]

Smotrov, 2024-01-03:
Well. Now it works! I've set 1450 VLAN on Web UI and it works now. 
Thank you all.

closed 2024-01-03 by Smotrov

Anime4000, 2024-01-04:
wow, this pure Ethernet, those 84 and 131 are noting..