Anime4000/RTL960x #233: DFP-34X-2C2  ONU State O5 rebooting

StreII, 2024-01-01:
[https://disk.yandex.ru/i/D0PR5AEjhgciZw](url)  video of the setup and defect.
What could be the reason?
and so with any firmware with the status 05, it reboots and does not get into the settings anymore until you pull out the optics.

Anime4000, 2024-01-02:
what Router/Switch do you use that run the ODI stick?

StreII, 2024-01-02:
Keenetic Peak (KN-2710) ,  GR-1000-AS-SFP

StreII, 2024-01-02:
there is an opinion that it is rebooting because of TR-069.
is there a function to disable the reboot after receiving the parameters from TR-069?

Anime4000, 2024-01-02:
ODI doesn't have TR-069

Anime4000, 2024-01-02:
if you from RU, try join the telegram here https://t.me/xpon_sfp_onu
they know what to do with RU ISP's

cuibapvh, 2024-01-02:
> https://disk.yandex.ru/i/D0PR5AEjhgciZw video of the setup and defect. What could be the reason? and so with any firmware with the status 05, it reboots and does not get into the settings anymore until you pull out the optics.

some ISPs try to upgrade the firmware via OMCI, the module doesn't understand what it is and causes 100% cpu then the modules freeze, check the latest firmware version from stock ONT and copy it into new module

StreII, 2024-01-03:
#

> > https://disk.yandex.ru/i/D0PR5AEjhgciZw video of the setup and defect. What could be the reason? and so with any firmware with the status 05, it reboots and does not get into the settings anymore until you pull out the optics.
> 
> some ISPs try to upgrade the firmware via OMCI, the module doesn't understand what it is and causes 100% cpu then the modules freeze, check the latest firmware version from stock ONT and copy it into new module I'm from Vietnam and my ISP can do it, my ISP updates their ONT regularly so I had to repack the original firmware of the module to get it working, without having to manually edit the new software version (this is quite annoying).

Do you also have ODI Realtek DFP-34X-2C2?

Anime4000, 2024-05-12:
try flash this firmware
[M110_sfp_ODI_240408_rel240513+hwtc-fix.zip](https://github.com/Anime4000/RTL960x/files/15287033/M110_sfp_ODI_240408_rel240513%2Bhwtc-fix.zip)

then edit OMCI including mac address, and leave MAC Key, it auto generate

hope not cause invalid mac key

CpServiceSpb, 2025-04-02:
> try flash this firmware [M110_sfp_ODI_240408_rel240513+hwtc-fix.zip](https://github.com/Anime4000/RTL960x/files/15287033/M110_sfp_ODI_240408_rel240513%2Bhwtc-fix.zip)
> 
> then edit OMCI including mac address, and leave MAC Key, it auto generate
> 
> hope not cause invalid mac key

Did you compile this FW ?

Anime4000, 2025-04-02:
> Did you compile this FW ?

Yes?

CpServiceSpb, 2025-04-18:
Can you change some .so or binaries files on iit ?