Anime4000/RTL960x #235: TP-Link BE900 Debug firmware to enable 2.5G SFP port (Link inside)

d5aqoep, 2024-01-04:
[image]

https://www.mediafire.com/file/8x1gbe6tped9mjs/BE900v1_1.1.3_240104_SFP_beta.zip/file

After updating to above debug beta firmware do the following (pass the debug commands in the web page given)

Go to http://192.168.0.1/webpages/debug.html

combo_debug stop
combo_debug set plus
combo_debug ipg_set combo10g 1_192bit

@Anime4000

Anime4000, 2024-01-04:
Nice, I add the firmware into repo, and link to this!

Good thing TP-Link listen to their customer, unlike Ubiquiti, Aruba and Fortinet (need CRS305 to make it 10G, cannot plug directly to save space)

Anime4000, 2024-01-04:
this is your post?
[image]

d5aqoep, 2024-01-04:
No. Guess they gave out the firmware today to whoever asked.

Anime4000, 2024-10-28:
I mark this issue as complete, TP-Link working to make GPON Stick support on BE800, BE900 and GE800
https://community.tp-link.com/en/home/forum/topic/711638

closed 2024-10-28 by Anime4000