Anime4000/RTL960x #236: lost ssh connection after firmware upgrade in ODI sfp in firmware M114_sfp_ODI_hybrid_221209.tar

jimpapi, 2024-01-13:
i upgraded my ODI stick in M114_sfp_ODI_hybrid_221209.tar. But i try to connect via SSH in port 22 and i can't. I tried to add rule in firewall but nothing. I can't connect from Kitty anymore. The firmware doesn't let me change OMCIInfo so i need ssh to pass the flash set commands.
Does anyone knows how to connect with ssh?
thank you

Anime4000, 2024-01-13:
I believe ODI use SHA1 SSH

```
ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 -oCiphers=+3des-cbc admin@192.168.1.1
```

I recommend connect ODI via Telnet instead

jimpapi, 2024-01-13:
i do connect if previous firmware. Only this firmware refuses/ I get the message connection refused. Something like port been closed. I tried firewall settings but again nothing. I ssh from windows with Kitty but yes from linux the ssh needs:
ssh -oKexAlgorithms=+diffie-hellman-group-sha1 -o HostKeyAlgorithms=+ssh-rsa -o PubkeyAcceptedAlgorithms=+ssh-rsa -c 3des-cbc admin@192.168.1.1

anyway i flashed the firmware [M110_sfp_ODI_220923.tar] and i had access again with ssh.

jimpapi, 2024-01-14:
ok Telnet works fine...ssh doesnt in this firmware version

jimpapi, 2024-01-14:
THANK YOU ANIME4000.

closed 2024-01-14 by jimpapi