Anime4000/RTL960x #240: Mac filtering on the SFP

ishanjain28, 2024-01-20:
Hey, 

My ISP annoyingly does not isolate customers. The SFP module gets all kinds of junk sent to it and I'd prefer to drop it all on the SFP instead of the router. Is adding mac filtering within the scope of this project ? sfp should only accept/frames from mac addresses on a allow list and drop every thing else.

Anime4000, 2024-01-21:
I think it's not possible,

it has basic `iptables` and `ebtables`, but those command are not permanent

perhaps try those

ishanjain28, 2024-01-22:
I tried with a ruleset like this, 

```

# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN
    link/loopback 00:00:00:00:80:05 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN qlen 1000
    link/ether 38:3a:21:28:b6:58 brd ff:ff:ff:ff:ff:ff
3: eth0.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
4: eth0.3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
5: nas0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
6: pon0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 00:00:00:01:00:02 brd ff:ff:ff:ff:ff:ff
7: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN
    link/ether 38:3a:21:28:b6:58 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br0

# ebtables -L
Bridge table: filter

Bridge chain: INPUT, entries: 4, policy: DROP
-s b4:8a:5f:24:3f:c1 -j ACCEPT # BNG / AC
-s a8:b8:e0:0:4a:d7 -j ACCEPT # RB5009

Bridge chain: FORWARD, entries: 4, policy: DROP
-s b4:8a:5f:24:3f:c1 -j ACCEPT
-s a8:b8:e0:0:4a:d7 -j ACCEPT

Bridge chain: OUTPUT, entries: 1, policy: ACCEPT
-j br_wan_out

Bridge chain: br_wan, entries: 0, policy: RETURN

Bridge chain: br_wan_out, entries: 0, policy: RETURN

Bridge chain: portmapping, entries: 0, policy: ACCEPT
```
And it does not work.

The SFP I have has a problem similar to some (non SFP) ONTs I have tried in the past. The rules added to `ebtables` only apply to traffic between SFP/ONT and the Router. It still allows all traffic from upstream. 

My objective here was to stop garbage sent by my ISP and that is not fixed by this modification. I hope there is a easy way to reset ONT(because I have locked myself out of it, :grimacing:) and I'll continue looking.

ishanjain28, 2024-01-22:
Oh now it makes sense.. 

`ebtables` is just for filtering traffic passing through linux bridges. [ref](https://superuser.com/questions/916221/prevent-traffic-to-mac-address-using-ebtables) 

I need to find some other way to filter traffic on `pon0` interface. I don't like the idea of adding it to a bridge and then filtering it.

ishanjain28, 2024-01-22:
It looks like this may not be doable on the stock firmware. I tried once again with `iptables` and it still allows all traffic from the ISP. This should be doable with `nftables` but `nftables` is not present on the device.

ishanjain28, 2024-02-16:
I do not believe this is possible on the SFP. 

1. `ebtables` is only for bridges and doesn't work here.
2. `iptables`(netfilter) can only filter traffic from router to sfp. It can not filter traffic from ISP. 
3. nftables is not available on the sfp.

I was discussing this with some people and the consensus so far is, the linux on sfp is probably just to provide a control interface. There is a high speed path from fiber to other interface that linux doesn't get to see or do any thing with!

closed 2024-06-20 by ishanjain28