Anime4000/RTL960x #249: Leox LXT-010S-H vlan problem

keczejo, 2024-02-28:
I bought leox LXT-010S-H as a replacement for ISP provided Zhone ZNID-GPON-2424-EU. I changed data to immitate Zhone device and i get O5 gpon state. My ISP is using VEIP and on top of that DHCP for address provisioning. Also i have multicast IPTV on a separate VLAN.

[image]

[image]

[image]

VLANs:
246 multicast IPTV
725 DHCP

At the first i didn't get any data comming from sfp interface. 
I contacted with Leox support and they provided me with firmware supporting VEIP. After firmware upgrade i started to recieve around 100Mb/s of data on sfp interface. It was multicast traffic on vlan246. I changed mac address on the router and enabled dhcp client on vlan 725, but there was no response. I also tried to bridge vlan 246 with ethernet port connected to STB. Unfortunately there was no television. 

Looking on PON statistics i see increasing amount of recieved data but number of data sent is 0 even though i am sending data to a proper VLAN on sfp interface. 

[image]

[image]

[image]

[image]

[image]

[image]

I tried this command `omcicli mib set 84 0x02 FwdOp 0x02 but it haven't changed anything.`

vuducdong, 2024-02-28:
Firmware that supports VEIP will have a V at the end, but I see your photo above is only V3.3.4L4. It should have been V3.3.4LV. Is there a mistake here or did Leox change the naming rules?

keczejo, 2024-02-28:
> 
> Firmware that supports VEIP will have a V at the end, but I see your photo above is only V3.3.4L4. It should have been V3.3.4LV. Is there a mistake here or did Leox change the naming rules?

My bad 
This screenshot was taken with firmware V3.3.4L4. On this software version there was no data on sfp port. I recieved firmware V3.3.4L5VT and V3.3.4L5V from leox support. On V3.3.4L5V OMCI_OLT_MODE=3 wasn't working, on V3.3.4L5VT OMCI_OLT_MODE=3 was working, i was receiving data on vlan 246, but data send by my router wasn't shown in pon statistics, exactly like on the screen shot.

ISP - > router is working
Router - > ISP isn't working

ToTheCLI, 2024-02-28:
DHCP (IPOE) sometimes needs MAC Address Cloning on SFP Interface, try to clone the MAC Address of original ONT WAN Interface.

keczejo, 2024-02-29:
> DHCP (IPOE) sometimes needs MAC Address Cloning on SFP Interface, try to clone the MAC Address of original ONT WAN Interface. 

I did that as a first thing. Weird thing is that my sfp module recieves DHCP requests but don't forward them to the optical interface.

keczejo, 2024-02-29:
> > > DHCP (IPOE) sometimes needs MAC Address Cloning on SFP Interface, try to clone the MAC Address of original ONT WAN Interface.
> > 
> > I did that as a first thing. Weird thing is that my sfp module recieves DHCP requests but don't forward them to the optical interface.
> 
> You can try it at the same time:
> 1, clone MAC ONT ---> MAC SFP GPON (Leox)
> 2, clone MAC port WAN Router (before running) ---> MAC port WAN router (new) (in this case, the sfp port)

Which command should i use to change ONT MAC?
I don't think it is the problem. I am sending requests and looks like the request is not send by the module to the OLT (bytes send counter is 0). I think sfp module doesn't permit packets from sfp to pon interface.

ROUTER -> SFP works 
SFP -> PON doesn't work

Anime4000, 2024-02-29:
have you tried to change FwdOp to 0x02 ? (Bridge, no investigation)

keczejo, 2024-03-01:
> have you tried to change FwdOp to 0x02 ? (Bridge, no investigation)

Yes i tried this solution. There was no change to the traffic. Should i do anything more after typing this command, like replugging fiber or something?

Anime4000, 2024-03-03:
it should work immediately, have you tried to conect Loex?

keczejo, 2024-03-04:
That was the first thing i did. They provided me different firmware versions and to try `omcicli mib set 84 0x02 FwdOp 0x02` . Can I try to install firmware from different SFP module with the same SoC on this one?

Anime4000, 2024-03-04:
RTL9601CI = V2801F, TWCGPON657 and many others, yea, but Leox stick is build differently even using same SoC, I would not recommend to flash it

if want to test, I advised you dump/backup whole SPI Flash, then you can try flash different fw

Mr1070, 2024-12-29:
i have had great luck with LeoLabs SFP .... i followed https://medium.com/@cyayon/configure-ont-leox-lxt-010h-d-for-orange-isp-4ef4b793ca7b and also took the advice to "flash set" the very minimum it is worked great