Anime4000/RTL960x #251: Can't access to uf instant anymore

NinoCasa, 2024-02-29:


NinoCasa, 2024-02-29:
I followed the instructions to swap unifi firmware with modified firmware that was published in stich86’s github repo https://github.com/stich86/UF-Instant-Mod
And when i switched back to unifi firmware, i can’t access anymore the uf instant with telnet or ssh, i got login incorrect or permission denied with ssh.
You can find the details of my procedure here https://github.com/stich86/UF-Instant-Mod/issues/20
Any help will be appreciated

Anime4000, 2024-02-29:
like this, you need access via UART and change boot partition to working one

NinoCasa, 2024-02-29:
Thank you for your response, i don't have any soldering skills but this is a good opportunity to learn, i will give it a try when i'll be on vacation 😄.
Until then, i ordered another one.

Anime4000, 2024-02-29:
alright, don't worry, usually most PON Stick have two bootable partition which is use for recovery to back to working firmware.

however, I would recommend to get ODI instead for the best and solve issue quickly.

still can adventure path to making UF Instant work

NinoCasa, 2024-02-29:
From my understanding the odi stick is for sc/upc but i jave sc/apc connector !!!

Anime4000, 2024-03-01:
you can ask seller to give APC, or use UPC-APC converter