Anime4000/RTL960x #252: Research to eliminate GPON SPF

eloydegen, 2024-02-29:
Not sure if this is has been attempted before, but I wondered if it is possible to use a non-GPON SPF and do all of the processing on the main router, such as the OMCI authentication, GEM encapsulation and the crypto on frames. Any ideas on this? I guess it would involve rewriting quite some software and maybe reverse engineer more binaries on the current GPON sticks.

eloydegen, 2024-02-29:
Found a kernel mailinglist mail: https://lore.kernel.org/netdev/20210605133113.j4gyjo4pnmxpxbqe@pali/

Seems not easy:

> > Adding GPON support properly into Linux is not an easy task, Linux would
> > probably need a subsystem with a complexity compared to cfg80211 + hostapd.
> 
> Yea... But maybe it could be easier to implement just "client part"
> (ONU/ONT) without "server part" (OLT).
>

Anime4000, 2024-02-29:
This mean using GPON Dumb Transceiver (eg: LTF7221-BH) and Host/Router do the OMCI ?

eloydegen, 2024-02-29:
> This mean using GPON Dumb Transceiver (eg: LTF7221-BH) and Host/Router do the OMCI ?

Yes.

Anime4000, 2024-02-29:
this would be interesting, imagine Mikrotik can install PON plugins and making CRS305 a versatile ONT