Anime4000/RTL960x #258: ISP Genexis Router has 4051 as VLAN but replacing it with V2802RH scan gives VLAN 1502

d5aqoep, 2024-03-13:
My ISP TataPlay Fiber has Alcatel OLT and gave me Genexis Router. In it's web GUI, I can see VLAN is 4051 for Internet and VLAN 4052 for voice. However, after cloning the GPON SN and PLOAM in Vsol V2802RH, the command `omcicli mib get 84` gives VLAN as 1502 and internet works only if I put 1502 as VLAN in the Vsol. 4051 does not work in it. 

Why are my VLANs changing if I use a 3rd party ONU? The thing is on VLAN 4051, IPv6 works correctly but on 1502, IPv6 has issues and does not assign non-temporary WAN IPv6. 

Can anything be done to make VSOL V2802RH use the VLAN 4051 according to my IPS's Router?
@Anime4000

Anime4000, 2024-03-13:
I haven't research on that VSOL, if interface is same like D-Link DPN-FX3060V

if ticking ✅ LAN 1 at WAN page
mean it will be untagged at LAN 1

d5aqoep, 2024-03-13:
> D-Link DPN-FX3060V

Yes the interface is very similar just different color theme. But I didn't get what you are saying.
It has 2 LAN ports. 1st port is 1Gbe and 2nd port is 2.5Gbe

Anime4000, 2024-03-13:
indeed,

when you [x] it mean VLAN will be untag, you need connect without any vlan

because VSOL has untag it

d5aqoep, 2024-03-13:
> indeed,
> 
> when you ✅ LAN X it mean VLAN will be untag, you need connect without any vlan
> 
> because VSOL has untag it

I mean try same 2nd port with no VLAN ?

Anime4000, 2024-03-13:
> I mean try same 2nd port with no VLAN ?

Yes