Anime4000/RTL960x #259: how to change Web GUI HTML5

qatar2030, 2024-03-14:
would you please advise me how to change Web GUI HTML5 on DFP-34X-2C2?
with full command 
and what firmware version can create triple vlan (vlan10-internet, vlan20-iptv and vlan30-telephone)
with regards

Anime4000, 2024-03-15:
try use IGD firmware, you can manually create VLAN

qatar2030, 2024-03-15:
link please

qatar2030, 2024-03-16:
ok if you don't where to find IGD firmware at least tell me how to change Web GUI HTML5 on DFP-34X-2C2 i think easy

Anime4000, 2024-03-16:
try this fw https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_hybrid_221209.tar

qatar2030, 2024-03-17:
> try this fw https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M114_sfp_ODI_hybrid_221209.tar

thank you so Mutch but  if I use this firmware I couldn't get internet connection when I create vlan10 for internet vlav30 for voip on it and vlan30  for voip I get connection up for only not internet but when I use M110_sfp_ODI_220923.tar firmware and create vlan10 on mikrotik I get internet connection