Anime4000/RTL960x #260: flash DFP-34X-2C2 with UART USB

qatar2030, 2024-03-14:
can someone advise me how to flash DFP-34X-2C2 with use UART USB coz I can't access via web Gui please?

Anime4000, 2024-03-15:
you need solder the RX and TX pins

aaokunev, 2024-03-29:
I had exactly the same problem.
1. I took my CH-341 programmer, switched it to TTL mode (set jumper to 2-3 pins)
2. I used flying probe for connect to tx/rx and power without soldering. See photos
3. I used a laboratory power supply to power the SFP module
4. Then used minicom on my mac to connect to SFP module via telnet: minicom -D /dev/tty.wchusbserial210 

[image]
[image]
[image]