Anime4000/RTL960x #261: LEOX LXT-010S-H failing to communicate

jvierstra, 2024-03-16:
I just got a LXT-010S-H and have it setup in a Media Converter connected to a Mikrotik router via Cat5. 

The device works, in the sense that I can log in via web and telnet, however, its gives me an ```O5``` status whenever its plugged into the fiber irrespective of whether a connection is made.

I dug into the OMCI logs at it appears that it is just losing communication with the OLT during messaging.
 
I cloned all of the OMCI data from the Calix ONT provided by my ISP using a UART hack to dump the device config.

Here is my config:

```
flash set OMCI_LOGFILE 1
flash set OMCI_DBGLVL 4
flash set ELAN_MAC_ADDR CCBE59XXXXXX
flash set OUI CCBE59
flash set PON_VENDOR_ID CXNK
flash set GPON_SN CXNK0053AC72
flash set GPON_PLOAM_PASSWD XXXXXXXXXX
flash set GPON_ONU_MODEL BVMHF00DRA
flash set HW_CWMP_MANUFACTURER Calix
flash set HW_CWMP_PRODUCTCLASS ONT
flash set HW_SERIAL_NO 261811025743
flash set HW_HWVER 3000267910
flash set OMCI_OLT_MODE 3
flash set OMCI_VENDOR_PRODUCT_CODE 20530
flash set OMCI_SW_VER1 12.2.10.5.6
flash set OMCI_SW_VER2 12.2.10.3.3
flash set OMCC_VER 177
reboot
```

After reboot I get the coveted ```O5``` status, albeit withe LOID Status in "Initial State". 

The strange thing is that I can't get a connection (the status page on the web interface of the PON shows no packet activity) and the communication with OLT seems to stall. Below is the output of the my ```omcilog.par``` file. It appears that only like ~25 messages are sent and some of them have this weird "UnknowME".

```
# cat omcilog.par
0x0c7c UnknowME 241
0x0c7c UnknowME 241
0x0c7d OltG Set(0x0000,Version=0x452030312e303000000000000000)
0x0c7d OltG SetRsp(0x0000,Result=0x0)
0x0c7e OltG Set(0x0000,OltVendorId=0x43414c58)
0x0c7e OltG SetRsp(0x0000,Result=0x0)
0x0c7f OltG Set(0x0000,EquipId=0x4537000000000000000000000000000000000000)
0x0c7f OltG SetRsp(0x0000,Result=0x0)
0x0de6 UnknowME 241
0x0de6 UnknowME 241
0x0de7 UnknowME 241
0x0de7 UnknowME 241
0x0de8 Ontg Set(0x0000,BatteryBack=0x01)
0x0de8 Ontg SetRsp(0x0000,Result=0x0)
0x0de9 Anig Set(0x8001,LowOpThreshold=0x3c,UppOpThreshold=0x0e,LowTranPowThreshold=0xe0,UppTranPowThreshold=0x14)
0x0de9 Anig SetRsp(0x8001,Result=0x0)
0x0dea GalEthProf Create(0x0001,MaxGemPayloadSize=0x065a)
0x0dea GalEthProf CreateRsp(0x0001,Result=0x0)
0x0deb LctUni Set(0x0000,AdminState=0x01)
0x0deb LctUni SetRsp(0x0000,Result=0x0)
0x0dec EthUni Set(0x0101,BridgedorIPInd=0x00)
0x0dec EthUni SetRsp(0x0101,Result=0x0)
0x0ded FecPmhd Create(0x8001,ThresholdData12Id=0xffff)
0x0ded FecPmhd CreateRsp(0x8001,Result=0x0)
0x0dee ZteSntp Delete(0x0000)
0x0dee ZteSntp DeleteRsp(0x0000,Result=0x0)
```

```
# diag gpon get onu-state
gpon get onu-state
ONU state: Operation State(O5)
```

I know something is strange because no VLAN information is sent back. 

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
#

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

Do you have any clues on why this is the case? Also, is there more debug tools on the LXT that I can use?

Thanks!

Anime4000, 2024-03-17:
you are not alone, here in Malaysia, one particular ISP using same as your OLT (CALIX)

from MA5671a to ODI to Leox stick, all same giving O5 without ME 84 and 171

Same issue as ME 241, not sure how to decrypt the Calix ME 241

ME 241 is vendor specific
[image]

qatar2030, 2024-03-17:
i have this sfp and it's work with mikrotik 5009
this is commands fork for me:
/etc/scripts/flash set GPON_SN HWTCCFD057AB
/etc/scripts/flash set GPON_PLOAM_PASSWD 1234567890
nv setenv sw_custom_version0 V5R021C10S146
nv setenv sw_custom_version1 V5R021C10S146
/etc/scripts/flash set HW_HWVER 16ED.A
/etc/scripts/flash set PON_VENDOR_ID HWTC
/etc/scripts/flash set GPON_ONU_MODEL HG8240T5
/etc/scripts/flash get OMCI_OLT_MODE
OMCI_OLT_MODE=0
/etc/scripts/flash set OMCI_OLT_MODE 1
/etc/scripts/flash get OMCI_FAKE_OK
OMCI_FAKE_OK=0
/etc/scripts/flash set OMCI_FAKE_OK 1