Anime4000/RTL960x #262: Security Assessment of ODI DFP-34X-2C2 Device

Dmitrizam, 2024-03-19:
Hello!

I am interested in the security of the ODI DFP-34X-2C2 device and would like to obtain more information about the security of this device, as well as the firmware it operates on.

Questions:

1. What security measures have been implemented in the ODI DFP-34X-2C2 device to ensure the protection of user data and network?
2. What firmware is used on the ODI DFP-34X-2C2 device? I would like to learn about the security of this firmware and its source.
3. Are there any known vulnerabilities or security issues that may affect the ODI DFP-34X-2C2 device? If so, what steps are being taken to address them?
I would appreciate any information or recommendations regarding the security of this device.

Thank you!

Anime4000, 2024-03-21:
One bug I found is, using same login cookie can make automatically login,

for example, using login cookie on Leox Stick that allow you to login automatically on ODI stick

the bug on boa I think,

also we been trying to compile simple hello world and run on the stick and none available, this could be issue RTL9601CI and D using modified MIPS, if we manage to build own toolchain, we can improve the stick to more robust and decompile the binary blobs

I not expert on this area, if anyone have spare time, would be nice to have

qatar2030, 2024-03-21:
you say to yourself you are not expert???? come on @Anime4000 you are a genius.

Anime4000, 2024-03-21:
well, I tried to compile own binary, it just can't run,

thus reverse engineering/decompile impossible if compiler toolchain not available.

first, we need build open source toolchain for RTL9601 CPU, so we can run own code

Dmitrizam, 2024-03-21:
Here, the comrades from China suggest buying a compiler toolchain from them.
[image]

Dmitrizam, 2024-03-29:
Hello, @Anime4000,

After analyzing the firmware of the ODI DFP-34X-2C2 device, I found that it uses Linux kernel version 2.6.30.9, which was released in 2009. Considering the age of this kernel version and possible unpatched vulnerabilities, I would like to update the system and build OpenWRT for this device to enhance its security level.

- What steps need to be taken to create a toolchain based on the existing firmware?
- Could you provide recommendations or point to resources that would assist in building OpenWRT for the ODI DFP-34X-2C2 module?

I am very interested in implementing this project and would appreciate any information that will help me in building OpenWRT for this SFP module.

Anime4000, 2024-03-29:
Yes, it use quite old kernel

second, RTL960x use heavily modified MIPS Architecture, no regular GCC compiler can compile and run

we have tried everything, still can't figure out how compile that RTL960x SoC understand.

If anyone can figure out RTL960x MIPS Architecture, it would be great, Reverse Engineer, improve, build, flash.

If this worked, can bridge automatically, if same VLAN exist on PPTP and VEIP, priority the PPTP, if none, bridge VEIP

arekm, 2024-03-29:
I assume known RSDKs were tested like https://github.com/qq431169079/rsdk-1.3.6-toolchain or https://github.com/frederic/rtl819x-toolchain/tree/master/toolchain/rsdk-1.5.5-5281-EB-2.6.30-0.9.30.3-110714 ?

crrodriguez, 2024-07-04:
Unless someone can secure access to the kernel modules of the XPON bits having a running toolchain is a wasted effort, but very much doable if one allocates time to do that. patches are posted online, base SOC stuff is as well.. but the interesting part isn't.

ramil-mux, 2024-11-14:
Hello @Dmitrizam !

I found an interesting detail. Could you please backup your configuration and find E8BDUSER_NAME?
I can use this login and password to log into my system