Anime4000/RTL960x #263: ODI 	DFP-34X-2C2 reboot loop when EPON link is lost

JeromeAstero, 2024-03-21:
Hello!
I'm encountering an issue with cyclic rebooting when losing connection with the OLT.
Device: Realtek **ODI DFP-34X-2C2** (V1.0-220923)
Router: **Keenetic Peak**
Link Type: **EPON**
The stick powers up for 30 seconds and then reboots. Sometimes, when I manage to access the web interface (PON tab), I see a block with GPON information, even though my connection is EPON. When there's a connection with the OLT, the stick works without any issues. It's possible that there are some hidden settings, but I'm unable to locate them.
[image]
[image]

vuducdong, 2024-03-21:
Please pay attention to mac and mackey, and enter any number for loid, loid pass, ploam pass (if it is empty)

vuducdong, 2024-03-22:
@JeromeAstero you tried fw version 220923FS?

and procedures after updating to the new version:

\\\\
flash_eraseall /dev/mtd3
reboot

JeromeAstero, 2024-03-24:
@vuducdong 
No, I haven't tried that firmware. Could you tell me more about it? Where can I get it? I haven't seen any firmware with the *FS suffix in this repository. And what is the difference between FS and the regular 220923 firmware?

vuducdong, 2024-03-24:
> @vuducdong 
> 
> No, I haven't tried that firmware. Could you tell me more about it? Where can I get it? I haven't seen any firmware with the *FS suffix in this repository. And what is the difference between FS and the regular 220923 firmware?

That's version fixed speed and vlan. Looks like it was removed from the repository.