Anime4000/RTL960x #264: DFP-34X-2C2 and UniFi UDM-SE No Connectivity

jmdomini, 2024-03-25:
Anyone have any experience using this GPON with a Ubiquity UDM-SE? Whether inserted directly into the USM-SE, a media converter or a Ciena and plugged into the WAN port on the UDM-SE I get no connectivity. If I plug that cable from the media convert or the Ciena into my laptop and assign and set my WAN IP there, everything works fine.

I have one VLAN (200) that I care about. I have tried both managing it on the GPON, as well as setting it to passthrough mode and setting it on the UDM-SE. Neither way works.

I can't figure what that UDM-SE is picking up that is breaking things that a directly connected laptop isn't. There's not much that's configurable on the UDM pretty much just IP, gateway, default subnet and default dscp marking if relevant.

My ISP definitely does not support removing their router from the mix so contacting their support is out of the question.

Anime4000, 2024-03-26:
first thing,

have you properly cloned from old ONT to the Stick? PLOAM Password, S/N, etc...

once you get O5, try check MIB 84, 131 and 171:
```
omcicli mib get 84
omcicli mib get 131
omcicli mib get 171
```

paste output here

jmdomini, 2024-03-26:
I believe I have everything cloned correctly. I get to 05 state and I'm assuming that my test with directly connecting my laptop to the GPON stick and successfully getting an internet connection wouldn't work if there was something wrong there.

Here's the result of the commands you asked for. I'm taking an educated guess here that 200 is the data VLAN because it's the only one that got me a connection in the laptop experiment when I plugged it into the VLAN settings on the GPON stick. I don't know what the rest of the data means though.

I also have an Alcatel G-010S-A stick I could try. But I can't figure out how to change the software version on it. There's a link on https://hack-gpon.org/ to instructions, but the instructions have been removed for some reason. Takedown request from Alcatel??

I'm dealing with an Alcatel OLT so it's very picky about the cloning details right.

```
`# m omcicli mob  ib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x1202
FilterTbl[0]: PRI 0,CFI 0, VID 200
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x120b
FilterTbl[0]: PRI 0,CFI 0, VID 200
FwdOp:  0x10
NumOfEntries: 1
=================================
# m omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x414c434c
EquipId:                     
Version: 
ToDInfo:
Sequence number of GEM superframe: 0x0
Timestamp: secs 0, nanosecs 0
=================================
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x101
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
AssociatedMePoint: 0x101
DscpToPbitMapping:
0x000000
0x249249
0x492492
0x6db6db
0x924924
0xb6db6d
0xdb6db6
0xffffff

Parsed DscpToPbitMapping:
dscp: 0 => pbit: 0
dscp: 1 => pbit: 0
dscp: 2 => pbit: 0
dscp: 3 => pbit: 0
dscp: 4 => pbit: 0
dscp: 5 => pbit: 0
dscp: 6 => pbit: 0
dscp: 7 => pbit: 0
[... 314 lines cut ...]
dscp: 42 => pbit: 5
dscp: 43 => pbit: 5
dscp: 44 => pbit: 5
dscp: 45 => pbit: 5
dscp: 46 => pbit: 5
dscp: 47 => pbit: 5
dscp: 48 => pbit: 6
dscp: 49 => pbit: 6
dscp: 50 => pbit: 6
dscp: 51 => pbit: 6
dscp: 52 => pbit: 6
dscp: 53 => pbit: 6
dscp: 54 => pbit: 6
dscp: 55 => pbit: 6
dscp: 56 => pbit: 7
dscp: 57 => pbit: 7
dscp: 58 => pbit: 7
dscp: 59 => pbit: 7
dscp: 60 => pbit: 7
dscp: 61 => pbit: 7
dscp: 62 => pbit: 7
dscp: 63 => pbit: 7
=================================
`
```

jmdomini, 2024-03-31:
Finally figured it out! First off don't bother trying to use this stick directly with the USM-SE, it's a lost cause and will simply loop boot. You'll need a media converter, a switch or in my case I put it in a Ciena 3903. The second part was changing the values below.

```
OMCI_CUSTOM_ME = 65792
OMCI_VEIP_SLOT_ID = 14
OMCI_VENDOR_PRODUCT_CODE = 0
```
Then I configured VLAN 200 on both the Ciena and the UDM-SE and I was good to go.

I don't really understand why I was able to connect a single device and seemingly have it work with these values apparently being incorrect. It was clear from studying the OMCI logs though that something was seriously broken.

Now I just hope after all this work that my ISP doesn't notice my "rogue" SFP and ban my port! ;-)

Anime4000, 2024-03-31:
> Now I just hope after all this work that my ISP doesn't notice my "rogue" SFP and ban my port! ;-)

ISP will notice no matter what, I got this from ISP when using the Stick:
[image]