Anime4000/RTL960x #265: Indihome OLT push configuration to enforce ODI running as router mode

nguyenthanhhong-tg, 2024-03-26:
Hi All,

Recently I got problem with Indihome on my ODI. Earlier, it worked fine in bridge mode and MikroTik dialed successfully for 3 years ago.

Now it seems Indihome did some changes on their OLT to enforce ODI running as router mode (as below image). OLT pushs PPPoE configuration and OID dialed directly on it.

I already triedwith SFU and HGU (M110_sfp_ODI_220304.tar, M114_sfp_ODI_Vlan_220414.tar, M114_sfp_ODI_hybrid_221209.tar) and got the same result.

Can someone let me know if we can disable or block ODI to dial directly on it so that my MikroTik can dial PPPoE?

Thank you,

[image]
[image]
[image]
[image]

Anime4000, 2024-03-26:
let's troubleshoot,

can you share the output:
```
omcicli mib get 84
omcicli mib get 131
omcicli mib get 171
```

nguyenthanhhong-tg, 2024-03-27:
Hi @Anime4000,

Below are output from my ODI. It is running firmware version V1.2.2-221209.

[image]
[image]
[image]
[image]
[image]
[image]

nguyenthanhhong-tg, 2024-03-27:
Hi @Anime4000,

You can see **NOTE: This wan is created by the olt!**, so that I can't delete this and create new one. Also can't modify current PPPoE account.

nguyenthanhhong-tg, 2024-03-27:
Let me try to switch image to M110_sfp_ODI_220304.tar and delete nas0_0. Last time I could delete.

nguyenthanhhong-tg, 2024-03-27:
I am running version M110_sfp_ODI_220304.tar and delete nas0_0 and nas0_1 and create new one as below.
So, theoretically, I will dial directly on sfp interface, no need create vlan then dial on that vlan, is it?
[image]

Anime4000, 2024-03-27:
Yes, try Admin Status: Disable, don't let OLT manage your port

nguyenthanhhong-tg, 2024-03-28:
I tried yesterday evening and it did not work.

1. First case: delete all auto configuration from OLT, then create interface **nas0_0** as bridge mode and tag VLAN 2483 on WAN interface of ODI.
2. Second case: delete all auto configuration from OLT on WAN interface, then create VLAN 2483 on MikroTik .

Anime4000, 2024-03-29:
I notice the ME 171, Entity Id 0x4101 some kind of special, the Me Point to 0x4### id

let's try override to use 0x101 (LAN 1)
```
omcicli mib set 171 0x4101 AssociatedMePoint 0x101
omcicli mib set 84 0x4101 FwdOp 0x02
omcicli mib set 84 0x101 FwdOp 0x02
```