Anime4000/RTL960x #266: Q: PON Splitter?

ztardik, 2024-03-27:
Not an issue, but a question.

Can I install a 1x2 PON splitter in front of everything? Will it cause signal loss or other issues? 

I would like to have a "failover" in case I need to reconnect the ISP equipment remotely. Found [this](https://www.aliexpress.com/item/33002357235.html?sku_id=67093696026&spm=a2g2w.stores.seller_list.13.341856f33saDYC) and the plan is to put the ISP equipment on one port and the DFP-34X-2C2 on the second port. In case there is an issue, I can just remotely energize the ISP equipment and shut down the DFP module. 
I have a backup link via LTE to do this.

Anime4000, 2024-03-27:
should be no issue, but you need 2nd PLOAM Password for 2nd Stick

ztardik, 2024-03-28:
Thanks. 
I posted the same question on [reddit](https://www.reddit.com/r/homelab/comments/1bourqr/pon_splitter/) and got some really in-depth responses I believe others can make use of. In essence, it's al about light levels - if there is enough light it can be done, if not, one can't do anything about it.

I guess I'll try to measure the light level in some way and see if it's a viable solution. From the response in reddit (u/PoisonWaffle3) I can guess that my line is at minimum 1x32 split as they provide a 300/150 link, but it is with overprovisioning of 0%. It can easily be a 1x64 and then I'm screwed.

> should be no issue, but you need 2nd PLOAM Password for 2nd Stick
Why do you think I need a second password? I guess it's something about the way I formed my post. I don't want and will never run both running at the same time.