Anime4000/RTL960x #268: O5 but omcicli mib get 84, 171 show no VLAN

rodfai, 2024-04-03:
ISP onu is fiberhome . I have already set RTK value.
Now I get O5 but that's it.   omcicli mib get 84 and 171 show nothing
VLAN in GUI is set to transparent.

[image]

but  diag   l2-table get entry address valid    show my VLAN
I set that VLAN in mikrotik but still cannot connect.  
What value or setting should I check it next?

Anime4000, 2024-04-03:
it appear that Fiberhome withheld MIB 84, have you tried to change OMCC ?

rodfai, 2024-04-03:
yep try all OMCC VER that I can choose but still no luck  :(

rodfai, 2024-04-21:
lot of HEC Error

[image]

I give up for now.  The guy from ISP said that they will change equipment to huawei soon .

panachoi, 2025-04-17:
My DFP-34X-2C2 stick is no longer returning any output from any omcicli at all, either. I know that it used to, because I have a log output (as outlined Docs/OMCI_CLI.md) from dumping it in December 2022.

I have not changed the firmware (V1.0-220923) or anything else (other than the LAN IP) since that time.

Any ideas ?