Anime4000/RTL960x #271: New firmware April 2024 for HSGQ / ODI DFP-34X-2C2

d5aqoep, 2024-04-08:
Yes here it is.

https://www.mediafire.com/file/zvpkk6t6rgywevz/M110_sfp_ODI_240407.tar/file

Anime4000, 2024-04-08:
I tried yesterday, contacted with the devs, still, few user can't use and had to stay with 220923

the issue has been told to devs, see how they handle PPTP vs VEIP

d5aqoep, 2024-04-08:
My issue is that VLAN tagging doesn't work with Asus or TP-Link router when VLAN is set in ODI

alexojegu, 2024-04-12:
@d5aqoep thanks, I'm trying it (PPTP seems enough for my ISP).

dylangerdaly, 2024-04-19:
Has anyone tried this newer firmware yet?

Anime4000, 2024-04-19:
someone have tried, many are fall back to old firmware

dylangerdaly, 2024-04-19:
Anyone know where the actual source of the firmware is?

I'd like to try but I want to make sure there's no signature checking or anything in the newer one.

chiragkrishna, 2024-04-20:
there is a new firmware [240408](https://www.mediafire.com/file/98qt78yw02netme/M110_sfp_ODI_240408.tar/file)

Vlan tagging is working, using with [TP-Link MC220L](https://www.tp-link.com/in/business-networking/accessory/mc220l/)
[image]

dylangerdaly, 2024-04-20:
VLAN Tagging works so we don't need the script anymore?

Anime4000, 2024-04-20:
the fix works with my ISP but not on others, still staying with 220923

[image]

fixing VLAN from PPTP and VEIP is not that simple, clashing VLAN on both can cause issue

dylangerdaly, 2024-04-20:
From a security perspective it'd be nice to know where these firmware versions are coming from and if they're legit.

I'm on the APC veriant of this device, and the VLAN Tagging Script causes packet loss, I'd like to try the newer version.

-------- Original Message --------

Doraemon309, 2024-04-28:
> there is a new firmware [240408](https://www.mediafire.com/file/98qt78yw02netme/M110_sfp_ODI_240408.tar/file)
> 
> Vlan tagging is working, using with [TP-Link MC220L](https://www.tp-link.com/in/business-networking/accessory/mc220l/) ![Screenshot from 2024-04-20 06-57-59](https://private-user-images.githubusercontent.com/2405186/324146016-7e24a65b-bb0f-4d6f-a778-c54d813b4e96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyODAyNzUsIm5iZiI6MTcxNDI3OTk3NSwicGF0aCI6Ii8yNDA1MTg2LzMyNDE0NjAxNi03ZTI0YTY1Yi1iYjBmLTRkNmYtYTc3OC1jNTRkODEzYjRlOTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjhUMDQ1MjU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTk1OTc0MDY5YmIwM2Y0NGMxNTllM2Y0NmMwNmUzNDU4NzM1NTM1MTQ2MjlkODc1MjkzYmIzZDM3OTljMjBhYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3A3MsABzl46wa1OJFWwCRLeuh8vitPoUPOu3Y4XveVk)

working!

Doraemon309, 2024-04-28:
> there is a new firmware [240408](https://www.mediafire.com/file/98qt78yw02netme/M110_sfp_ODI_240408.tar/file)
>
> Vlan tagging is working, using with [TP-Link [...]

Working on mikrotik and openwrt!
It seems like some places or OSs don't support the new firmware well!

chesskuo, 2024-09-15:
Hello,  I got some problem about 240408 firmware.

When I upgrade to this version, and using the correct settings connecting to the OLT.
The stick would downgrade to 220923, it seems to trigger some bug and auto rollback to the last version.

If I using a wrong config makes it connection failed to OLT, it can boot to new firmware successfully.

BUT, if I already crash and downgrad to the old version, even through I use the wrong config.
It still keep on the old version (220923) and would not auto change to the new firmware when I reboot.

kelt84, 2024-10-10:
Hello everyone, tell me, maybe someone knows how to solve this issue, I bought a DFP-34X-2C2 and reflashed it with all available firmware, the situation is as follows, authentication with the provider is carried out by the serial number of the device, when entering the serial number and then saving, it asks to enter the PLOAM password, but since authentication is carried out by the serial number, the PLOAM field must be left empty, how can this be done?

Anime4000, 2024-10-10:
either you can fill with 10 zeros or download the config, edit, and upload

kelt84, 2024-10-10:
I tried the option with 10 zeros, I would be grateful if you could tell me how to download the config file?

kelt84, 2024-10-10:
I edited the config file, it worked, although identification does not pass, thanks for the assistance, I will continue to figure it out.

giovannybm, 2024-11-01:
greetings, I have updated to this firmware but when I want to install any other image I get error, invalid file.

How can I install a previous firmware?

[image]

ifplusor, 2024-11-03:
The IGMP Snooping on 240408 isn’t working well. After the downstream joins the group using IGMP v2, it doesn’t receive any data.

dongleactive, 2024-11-25:
> Yes here it is.
> 
> https://www.mediafire.com/file/zvpkk6t6rgywevz/M110_sfp_ODI_240407.tar/file
> 
> @Anime4000 you can add it to repository after confirmation that it works and has fixed VLAN tagging issue. Mediafire can delete it after a period of inactivity.

> I tried yesterday, contacted with the devs, still, few user can't use and had to stay with 220923
> 
> the issue has been told to devs, see how they handle PPTP vs VEIP

i'm too, How do I get back to the old firmware? I reinstalled the old version and it showed the error, the new version is not blocked by mac router, olt can't see the router's mac.

alexeymx, 2024-12-21:
guys i have this stick but it asks me for loid and loid password. my carrier is telmex and they do not use any loid auth

Doraemon309, 2024-12-22:
> guys i have this stick but it asks me for loid and loid password. my carrier is telmex and they do not use any loid auth

You must contact the ISP directly to ask for a reasonable solution!