Anime4000/RTL960x #272: MEO Portugal / UFiber NanoG / Help Identifying my GPON SFP+ Module

dylangerdaly, 2024-04-10:
Hello,

I'm attempting to replace my ISP's ONT with an SFP+ Module I got from Aliexpress

This is the listing: https://www.aliexpress.com/item/4000537015384.html

> E/GXPON SFP ONU Stick With MAC SC Connector DDM pon module 1490/1330nm 1.25/2.5G XPON/EPON/GPON(1.244Gbps/2.55G)802.3ah E/GXPON

I've been able to successfully replace it with a Ubquiti UFiber NanoG, only changing the Serial to match the original ONT works, however download is limited to ~250Mbps and upload is limited to 1Mbps, I assume this is because of a laser issue.

Ultimately I'd like to use the SFP+ Module above however I'm not sure what version of the modified firmware I should install, setting the GPON Serial Number results in O0 (WRONG) from the OLT.

Any help is greatly appreciated 

Screenshot of NanoG
[image]

Anime4000, 2024-04-11:
have you tried change `OMCI_TM_OPT` `0` `1` `2` (default) ?

dylangerdaly, 2024-04-19:
MEO works with the SFP+

closed 2024-04-19 by dylangerdaly