Anime4000/RTL960x #274: VLAN not working after reboot

serniko97, 2024-04-14:
Hi, I have an ODI DFP-34G-2C2 with firmware v1.0-220923. I configured everything following this guide and it works very well with my ISP, thank you very much for this repository.

The only issue I encountered is with VLANs:
everything works perfectly if I set the ONT in transparent mode and set the VLAN tag 835 directly on the router, but if I set the VLAN on the ONT and remove the VLAN from my router it only works until the ONT is rebooted, then to make it work again I have to re-apply the VLAN setting on the ONT.

Anime4000, 2024-04-16:
can you telnet and show the results of:
```
omcicli mib get 84
omcicli mib get 171
omcicli mib get 256
```

serniko97, 2025-03-01:
Hi @Anime4000, I'm very sorry for replying after almost 1 year.

I checked what you asked and this is the output you requested, it is the same before (while it's working) and after rebooting (when it stops working):
```
BusyBox v1.12.4 (2022-09-23 19:33:44 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x01
FwdOp:  0x01
NumOfEntries: 0
=================================
=================================
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 838
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 837
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 835
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 836
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x06
FwdOp:  0x15
NumOfEntries: 0
=================================
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 256
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Ontg
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x00
VID: HWTC
Version: 6A5.A
SerialNum: HWTCe928256c
TraffMgtOpt: 2
AtmCCOpt: 0
BatteryBack: 1
AdminState: 0
OpState: 0
OnuSurvivalTime: 0
LogicalOnuID: 
LogicalPassword: 
CredentialsStatus: 0
ExtendedTcLayerOptions: 0x0
OntState: 1
=================================
# 
```