Anime4000/RTL960x #276: MIB 257 Equipment ID encoding

Gruni22, 2024-04-19:
Hi,

first of all, I want to thank you for all your effort!!!
I have the DFP-34X-2C2 GPON Stick and managed to get it on ONU State O3 with cloning data.
In the past I already managed to get the FS.com GPON and ZTE F6005 working.
The FS.com cannot sync with 2,5 Gbit because my SFP+ Card does not bring it up.
The ZTE F6005 is a nice solution, but it's not SFP ;-)

Now my issue:
No matter which firmware I try, they seem to have problems of proper HEX Encoding of MIB 257 data.
What I need is exactly that (same output from working devices) (shortened output):
`EqtID                     = 5f 5f 5f 5f 5f 5f 5f 5f 5f 5f 47 2d 30 31 30 47 2d 51 5f 5f 
OMCCVer                   = 160
NumOfPriQ                 = 33
NumOfScheduler            = 9
NumOfGemPort              = 33
QosCfgFlexibility         = 1`

What I get is with DFP-34X-2C2 that (shortened output):
`EqtID                     : ___________G-010G-Q__  # should be Hex encoded
OMCCVer                   : 0xa0 #HEX for 160 - should be OK
NumOfPriQ                 : 128 # No Idea how to change or relevant
NumOfScheduler            : 9 # No Idea how to change or relevant
NumOfGemPort              : 64 # No Idea how to change or relevant
QosCfgFlexibility             : 0x003b # No Idea how to change or relevant`

Do you have any idea how to change those values?

Kind Regards,
Matthias

Anime4000, 2024-04-19:
if you query from `omcicli mib get <id>` most the value is Decimal by default, when in Hex, always start with `0x`

maybe you can try `omcicli mib set 257 OMCCVer 0xa0` see what happen

syntax: `omcicli mib set <mib_id> <instance_name> <value>`

Gruni22, 2024-04-19:
Thanks for response!
If I enter:
`omcicli mib set 257 0x0000 EqtID G-010G-Q`
I receive:
`EqtID: 0x472d303130472d51000000000000000000000000`

If I enter:
`omcicli mib set 257 0x0000 EqtID 0x5f5f5f5f5f5f5f5f5f5f472d303130472d515f5f`
I receive:
`EqtID: 0x5f5f5f5f5f5f5f5f5f5�`

Did also the test with:
`omcicli mib set 257 0x0000 OMCCVer 0xa0`
`omcicli mib set 257 0x0000 OMCCVer 0xaa`
Where both values are interpreted correctly.

ToTheCLI, 2024-04-26:
> Thanks for response! If I enter: `omcicli mib set 257 0x0000 EqtID G-010G-Q` I receive: `EqtID: 0x472d303130472d51000000000000000000000000`
> 
> If I enter: `omcicli mib set 257 0x0000 EqtID 0x5f5f5f5f5f5f5f5f5f5f472d303130472d515f5f` I receive: `EqtID: 0x5f5f5f5f5f5f5f5f5f5�`
> 
> `omcicli mib set 257 0x0000 EqtID __________G-010G-Q_` `EqtID: 0x5f5f5f5f5f5f5f5f5f5f472d303130472d515f00`
> 
> `omcicli mib set 257 0x0000 EqtID __________G-010G-Q__` `EqtID: __________G-010G-Q__`
> 
> Did also the test with: `omcicli mib set 257 0x0000 OMCCVer 0xa0` `omcicli mib set 257 0x0000 OMCCVer 0xaa` Where both values are interpreted correctly.

Did you find a solution?

Gruni22, 2024-04-26:
No. 
Ich contacted the supplier for a firmware update. 
They said it should work, but it does not.