Anime4000/RTL960x #278: ODI SFP DFP-34X-2C2 Does not start after applying new firmware

lucioborges, 2024-04-23:
Checking the boot via the UART Serial console, the system does not boot.
Is there a way to perform a reimage via xmodem?
I can access U-boot

9601D# ?
?               - alias for 'help'
base            - print or set address offset
bdinfo          - print Board Info structure
boot            - boot default, i.e., run 'bootcmd'
bootd           - boot default, i.e., run 'bootcmd'
bootm           - boot application image from memory
bootp           - boot image via network using BOOTP/TFTP protocol
btg             - bus traffic generator sub-system
cmp             - memory compare
coninfo         - print console devices and information
cp              - memory copy
crc32           - checksum calculation
echo            - echo args to console
editenv         - edit environment variable
env             - environment handling commands
exit            - exit script
false           - do nothing, unsuccessfully
go              - start application at address 'addr'
help            - print command description/usage
iminfo          - print header information for application image
imxtract        - extract a part of a multi-image
itest           - return true/false on integer compare
loadb           - load binary file over serial line (kermit mode)
loads           - load S-Record file over serial line
loady           - load binary file over serial line (ymodem mode)
loop            - infinite loop on address range
md              - memory display
mdram_test      - mdram_test   - do DRAM test.
mflash_test     - mflash_test   - do flash test.
mm              - memory modify (auto-incrementing address)
mtest           - simple RAM read/write test
mw              - memory write (fill)
nm              - memory modify (constant address)
printenv        - print environment variables
reset           - Perform RESET of the CPU
reset_all       - Perform whole chip RESET of the CPU
reset_sw        - Perform software RESET
run             - run commands in an environment variable
saveenv         - save environment variables to persistent storage
setenv          - set environment variables
sf              - SPI flash sub-system
showvar         - print local hushshell variables
sleep           - delay execution for some time
source          - run script from memory
test            - minimal test like /bin/sh
tftp            - boot image via network using TFTP protocol
tftpboot        - boot image via network using TFTP protocol
true            - do nothing, successfully
upimgtar        - update kernel and rootfs by tar format on luna platform
upvmimg         - update kernel and rootfs by vmimg format on luna platform
version         - print monitor, compiler and linker version
xf              - xmodem to flash
9601D#

lucioborges, 2024-05-01:
MX25L6433F flash memory dump?

Anime4000, 2024-05-02:
try `setenv sw_active 0` and `setenv sw_commit 0` ?

check `printenv sw_active` first see value is 0 or 1,

if output 0, set to 1,
if output 1, set to 0

lucioborges, 2024-05-02:
I've already tried changing the system from 0 to 1 without success. I've already bought another SFP that should arrive this week!
For testing, I removed the MX25L6433F memory and am looking for an image of this memory to rewrite.
Thanks for the support. If I can solve it, I'll share it here.

Anime4000, 2024-05-03:
Alright, if you managed to fix the MX25L6433F SPi Memory, the rootfs partition, share it, I'll link this from README.md

lucioborges, 2024-05-09:
I wasn't successful in rewriting the chip, but I think I burned the power circuit of the old SFP by using my soldering iron on it, which isn't ideal. But I ended up copying a working image from my SFP OK, which is actually working.
But to rewrite the Chip you have to remove it from the board and use a CH341A reader/writer

If you need it, here is an image of the memory
https://drive.google.com/drive/folders/1j3f9sfIdDUlwOiKLeQtyCpPKvczfsN36?usp=drive_link

lehoangnb, 2024-05-10:
> I wasn't successful in rewriting the chip, but I think I burned the power circuit of the old SFP by using my soldering iron on it, which isn't ideal. But I ended up copying a working image from my [...]

Dont need to remove the flash ic, it can be programmed directly on the board!

lucioborges, 2024-05-10:
With the memory soldered on the board, I was unable to read and write.

lehoangnb, 2024-05-10:
> With the memory soldered on the board, I was unable to read and write.

Maybe programmer not produce enough power

Anime4000, 2024-05-10:
I have tried read SPI Flash direct from the board, thing is, the 3.3v is powering the stick when I start reading from CH341A

That time, my PON Stick still connected via UART, then I read the SPI Flash, cause PON Stick to power up and start booting

you can try read one, and verify multiple time, it will mismatch, so I end up need remove the SPI Flash off the board which is extra work to do

lehoangnb, 2024-05-11:
> I have tried read SPI Flash direct from the board, thing is, the 3.3v is powering the stick when I start reading from CH341A
>
> That time, my PON Stick still connected via UART, then I read the SPI [...]

My programmer have power control feature, it's only enable 3.3V on operation (read, write, ...) it will block sfp from booting (maybe cpu cannot read spi flash and stop operation), I've  read multi time, all CRC checksums are the same.
For other programer, you can try to connect power only when read/write,etc...

Anime4000, 2024-05-11:
your Programmer much better than  CH341a that I have.

May I know what programmer are you using?

does Programmer software has automatic SFP EEPROM Checksums?

lehoangnb, 2024-05-11:
> your Programmer much better than CH341a that I have.
> 
> May I know what programmer are you using?
> 
> does Programmer software has automatic SFP EEPROM Checksums?

It's [Skypro II](http://www.coright.com/product/showproduct.php?lang=en&id=55), like other programer, its only do checksum for whole dump file, for SFP, its have sperate crc for config and info( PN, SN, mf date,...) (0x3f for config, 0x5f for info)  and programer cannot calculate it automaticaly.

Anime4000, 2024-05-11:
> > your Programmer much better than CH341a that I have.
> > May I know what programmer are you using?
> > does Programmer software has automatic SFP EEPROM Checksums?
> 
> It's [Skypro II](http://www.coright.com/product/showproduct.php?lang=en&id=55), like other programer, its only do checksum for whole dump file, for SFP, its have separate crc for config and info( PN, SN, mf date,...) (0x3f for config, 0x5f for info) and programer cannot calculate it automaticaly.

I see, this mean still need to use `SFPTotal Wizard` to for automatic SFP checksum.

I guess I try with my existing CH341a and [IMSProg](https://github.com/bigbigmdm/IMSProg)

after you show the EEPROM, now I confident and bought this for quick programming
[image]