Anime4000/RTL960x #279: Different VLAN in ISP provided ONU compared to V2802RH

d5aqoep, 2024-04-25:
I have already raised this issue before. My VLAN shown in ISP provided ONU is 3350. However, any other ONU I replace after changing GPON SN and PLOAM, it shows 1502. As recommended, I tried using LAN port 1 (1Gig) of V2802RH but the issue persists. Same issue with TP-Link ONU too. 

My ISP is uisng Alcatel OLT and is this some bug that VLAN changes or OLT expects some specifit OMCI information to enable their standard 3350 VLAN?

I have IPv6 related issues on 1502 VLAN, however no such issue exist when I use ISP provided ONU.

Anime4000, 2024-04-26:
I guess ME 171, PON Side and LAN Side

d5aqoep, 2024-05-02:
@Anime4000 I didn't get you. Can you help?

Anime4000, 2024-05-02:
paste your ME 171 here

d5aqoep, 2024-05-02:
How do I do that with telnet?

d5aqoep, 2024-05-02:
AP#omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

ExtVlanTagOperCfgData

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

=================================

EntityId: 0x101

AssociationType: 2

ReceivedFrameVlanTagOperTableMaxSize: 0

InputTPID: 0x8100

OutputTPID: 0x8100

DsMode: 0

ReceivedFrameVlanTaggingOperTable

INDEX 0

Filter Outer   : PRI 8,VID 4051, TPID 5

Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 9,VID 1502, TPID 3

INDEX 1

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 8,VID 4051, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 8,VID 1502, TPID 2

INDEX 2

Filter Outer   : PRI 8,VID 0, TPID 5

Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 9,VID 1502, TPID 3

INDEX 3

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 8,VID 0, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 8,VID 1502, TPID 2

INDEX 4

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 15,VID 4051, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 0

Treatment Inner   : PRI 0,VID 1502, TPID 2

INDEX 5

Filter Outer   : PRI 14,VID 4096, TPID 5

Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 3

INDEX 6

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 2

AssociatedMePoint: 0x101

DscpToPbitMapping:

        0x000000

        0x249249

        0x492492

        0x6db6db

        0x924924

        0xb6db6d

        0xdb6db6

        0xffffff

Parsed DscpToPbitMapping:

dscp: 0 => pbit: 0

dscp: 1 => pbit: 0

dscp: 2 => pbit: 0

dscp: 3 => pbit: 0

dscp: 4 => pbit: 0

dscp: 5 => pbit: 0

dscp: 6 => pbit: 0

dscp: 7 => pbit: 0

dscp: 8 => pbit: 1

dscp: 9 => pbit: 1

dscp: 10 => pbit: 1

dscp: 11 => pbit: 1

dscp: 12 => pbit: 1

dscp: 13 => pbit: 1

dscp: 14 => pbit: 1

dscp: 15 => pbit: 1

dscp: 16 => pbit: 2

dscp: 17 => pbit: 2

dscp: 18 => pbit: 2

dscp: 19 => pbit: 2

dscp: 20 => pbit: 2

dscp: 21 => pbit: 2

dscp: 22 => pbit: 2

dscp: 23 => pbit: 2

dscp: 24 => pbit: 3

dscp: 25 => pbit: 3

dscp: 26 => pbit: 3

dscp: 27 => pbit: 3

dscp: 28 => pbit: 3

dscp: 29 => pbit: 3

dscp: 30 => pbit: 3

dscp: 31 => pbit: 3

dscp: 32 => pbit: 4

dscp: 33 => pbit: 4

dscp: 34 => pbit: 4

dscp: 35 => pbit: 4

dscp: 36 => pbit: 4

dscp: 37 => pbit: 4

dscp: 38 => pbit: 4

dscp: 39 => pbit: 4

dscp: 40 => pbit: 5

dscp: 41 => pbit: 5

dscp: 42 => pbit: 5

dscp: 43 => pbit: 5

dscp: 44 => pbit: 5

dscp: 45 => pbit: 5

dscp: 46 => pbit: 5

dscp: 47 => pbit: 5

dscp: 48 => pbit: 6

dscp: 49 => pbit: 6

dscp: 50 => pbit: 6

dscp: 51 => pbit: 6

dscp: 52 => pbit: 6

dscp: 53 => pbit: 6

dscp: 54 => pbit: 6

dscp: 55 => pbit: 6

dscp: 56 => pbit: 7

dscp: 57 => pbit: 7

dscp: 58 => pbit: 7

dscp: 59 => pbit: 7

dscp: 60 => pbit: 7

dscp: 61 => pbit: 7

dscp: 62 => pbit: 7

dscp: 63 => pbit: 7

=================================

=================================

EntityId: 0x601

AssociationType: 10

ReceivedFrameVlanTagOperTableMaxSize: 0

InputTPID: 0x8100

OutputTPID: 0x8100

DsMode: 0

ReceivedFrameVlanTaggingOperTable

INDEX 0

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 8,VID 4052, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 8,VID 1943, TPID 2

INDEX 1

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 0, TPID 2

INDEX 2

Filter Outer   : PRI 14,VID 4096, TPID 5

Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 3

INDEX 3

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 2

AssociatedMePoint: 0x601

DscpToPbitMapping:

        0x000000

        0x249249

        0x492492

        0x6db6db

        0x924924

        0xb6db6d

        0xdb6db6

        0xffffff

Parsed DscpToPbitMapping:

dscp: 0 => pbit: 0

dscp: 1 => pbit: 0

dscp: 2 => pbit: 0

dscp: 3 => pbit: 0

dscp: 4 => pbit: 0

dscp: 5 => pbit: 0

dscp: 6 => pbit: 0

dscp: 7 => pbit: 0

dscp: 8 => pbit: 1

dscp: 9 => pbit: 1

dscp: 10 => pbit: 1

dscp: 11 => pbit: 1

dscp: 12 => pbit: 1

dscp: 13 => pbit: 1

dscp: 14 => pbit: 1

dscp: 15 => pbit: 1

dscp: 16 => pbit: 2

dscp: 17 => pbit: 2

dscp: 18 => pbit: 2

dscp: 19 => pbit: 2

dscp: 20 => pbit: 2

dscp: 21 => pbit: 2

dscp: 22 => pbit: 2

dscp: 23 => pbit: 2

dscp: 24 => pbit: 3

dscp: 25 => pbit: 3

dscp: 26 => pbit: 3

dscp: 27 => pbit: 3

dscp: 28 => pbit: 3

dscp: 29 => pbit: 3

dscp: 30 => pbit: 3

dscp: 31 => pbit: 3

dscp: 32 => pbit: 4

dscp: 33 => pbit: 4

dscp: 34 => pbit: 4

dscp: 35 => pbit: 4

dscp: 36 => pbit: 4

dscp: 37 => pbit: 4

dscp: 38 => pbit: 4

dscp: 39 => pbit: 4

dscp: 40 => pbit: 5

dscp: 41 => pbit: 5

dscp: 42 => pbit: 5

dscp: 43 => pbit: 5

dscp: 44 => pbit: 5

dscp: 45 => pbit: 5

dscp: 46 => pbit: 5

dscp: 47 => pbit: 5

dscp: 48 => pbit: 6

dscp: 49 => pbit: 6

dscp: 50 => pbit: 6

dscp: 51 => pbit: 6

dscp: 52 => pbit: 6

dscp: 53 => pbit: 6

dscp: 54 => pbit: 6

dscp: 55 => pbit: 6

dscp: 56 => pbit: 7

dscp: 57 => pbit: 7

dscp: 58 => pbit: 7

dscp: 59 => pbit: 7

dscp: 60 => pbit: 7

dscp: 61 => pbit: 7

dscp: 62 => pbit: 7

dscp: 63 => pbit: 7

=================================

AP#

d5aqoep, 2024-05-03:
@Anime4000 

flash set LAN_SDS_MODE 1

this command didn’t work after commit and reboot.

d5aqoep, 2024-05-05:
@Anime4000 Hello are you there>
you can help me over Telegram. My telegram username is @d5aqoep

d5aqoep, 2024-05-20:
@Anime4000 any solution for this problem?