Anime4000/RTL960x #280: change DFP-34X-2C2 logo

qatar2030, 2024-04-26:
[image]

i would like to change logo on DFP-34X-2C2
how to do that please

Anime4000, 2024-04-26:
use the emulator file that I made:
https://github.com/Anime4000/RTL960x/tree/main/Tools/emulator

billbarni, 2024-10-18:
I've run the emulator, copied the files to the 'custom' directory, closed the emulator so it builds the firmware... but when I upload the firmware via GUI, nothing happens.
It's mandatory to use netcat for this? What am I doing wrong?
I see a message when emulator is being closed "Skip squashfs-root as not wanted in firmware package", but there lies the files I want added...

Anime4000, 2024-10-19:
try run in Private Browsing to avoid cache

the custom folder is mirror/overlay of root squahfs

need place the file accurately