Anime4000/RTL960x #282: O5 but no ONI ID and no connection

erdemefe90, 2024-04-28:
Hello Everyone,
I'm from Turkey and I use turktelekom as ISS. they provide me GPON ONT which is ZTE ZXHN F601 and does not supply any information about changing their ONT. I have a Mikrotik RB4011 router and eth1 tagged the vlan35 and my pppoe client uses vlan35 as interface. My connection speed is 1000/100mbps
i am tring to use DFP-34X-2C2 as GPON ONT like everbody is here. But my SFP ONT brand is not ODI it is Xicom I bought it from aliexpress so I'm afraid of changing its firmware because all firmwares tested on ODI. My version is "V1.0-220923"

i have connection to ZTE ONT via http and telnet and I can quering to ME codes and it have "setmac" command for some configs. I collected all information from ZTE. they are below.

I have always got O5 state but no ONU ID and LOID status is Initial Status. I have no connection to internet. What is missing or how can I debug the auth state?

As I understand setmac commands defination below. But Im not sure
setmac 2 512 gives device serial number
setmac 2 768  gives OUI
setmac 2 2176 PON vendor ID (first 4 char of gpon serial)
setmac 2 2177 GPON serial number (last 8 char of gpon serial)
setmac 2 2178 PLOAM password.

Loid and loid password obviosly seems in ME 256 query

## ZTE setmac commands
```
setmac 2 512			
ZTELQDLP3H05829
setmac 2 768 			
3CA7AE
setmac 2 256			
3c a7 ae 05 8e e4
setmac 2 2176 			
ZTEG
setmac 2 2177			
4D6F211C
setmac 2 2178			
G4D6F211C
```
## ZTE OMCI ME queries
```
##################################
MIB INFO:
         ME CLASS: 7
         DB NAME: soft_image, DBHandle: 14
##################################
<-----MeID[ 0x0000,0 ], Addr[ 0x19cf31]----->
          Version:V7.0.10P1N37
     Is committed:00 
        Is active:00 
         Is valid:01 
<-----MeID[ 0x0001,1 ], Addr[ 0x19cf51]----->
          Version:V7.0.10P1N37
     Is committed:01 
        Is active:01 
         Is valid:01 
---------------------------------------------------------------------
Total Record(s) num: 2

##################################
MIB INFO:
         ME CLASS: 131
         DB NAME: olt_g, DBHandle: 32
##################################
<-----MeID[ 0x0000,0 ], Addr[ 0x19d1d9]----->
         Vendorid:48 57 54 43 
      EquipmentID:00 00 00 00 00 00 00 00 00 00 
                    00 00 00 00 00 00 00 00 00 00 
          Version:31 30 00 00 00 00 00 00 00 00 
                    00 00 00 00 
        TimeofDay:00 00 00 00 00 00 00 00 00 00 
                    00 00 00 00 
---------------------------------------------------------------------
Total Record(s) num: 1

##################################
MIB INFO:
         ME CLASS: 84
         DB NAME: vlan_filter, DBHandle: 69
##################################
<-----MeID[ 0x01ff,511 ], Addr[ 0x1e96c1]----->
 vlan_filter_list:00 00 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 
      forward_opr:04 
           number:00 
<-----MeID[ 0x0003,3 ], Addr[ 0x289a21]----->
 vlan_filter_list:00 4b 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 
      forward_opr:10 
           number:01 
<-----MeID[ 0x0004,4 ], Addr[ 0x289909]----->
 vlan_filter_list:06 40 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 00 00 00 00 00 00 00 00 
                          00 00 
      forward_opr:10 
           number:01 
---------------------------------------------------------------------
Total Record(s) num: 3

##################################
MIB INFO:
         ME CLASS: 256
         DB NAME: ont_g, DBHandle: 22
##################################
<-----MeID[ 0x0000,0 ], Addr[ 0x19ce51]----->
         Vendorid:5a 54 45 47 
          Version:56 37 2e 30 00 00 00 00 00 00 
                    00 00 00 00 
     Serialnumber:5a 54 45 47 4d 6f 21 1c
        TrafficMO:02 
            VP/VC:00 
    Batterybackup:01 
          adstate:00 
          opstate:00 
      survialtime:00 
     logicalONUID:31 32 33 34 35 36 37 38 39 00 
                    00 00 00 00 00 00 00 00 00 00 
                    00 00 00 00 
  logicalPassword:31 32 33 34 35 36 00 00 00 00 
                    00 00 
credentialsStatus:00 
---------------------------------------------------------------------
Total Record(s) num: 1

##################################
MIB INFO:
         ME CLASS: 257
         DB NAME: ont2_g, DBHandle: 23
##################################
<-----MeID[ 0x0000,0 ], Addr[ 0x19cea9]----->
      EquipmentID:46 36 30 31 56 37 2e 30 00 00 
                    00 00 00 00 00 00 00 00 00 00 
      OmccVersion:b2 
VendoPproductCode:00 00 
SecurityCapability:01 
     SecurityMode:01 
TotalPriorityQueueNumber:00 38 
TotalTrafficSchedulerNumber:07 
       Deprecated:01 
       GemPortNum:01 00 
        SysUpTime:00 00 00 00 
    Concapability:00 7f 
        CurConMod:00 
        Qosconfig:00 3e 
          PQscale:00 01 
---------------------------------------------------------------------
Total Record(s) num: 1
```

i applied all information to DFP-34X-2C2 but there is no connection. They are below.
## DFP-34X-2C2 config
```
# flash get GPON_SN
GPON_SN=ZTEG4D6F211C
# flash get GPON_PLOAM_PASSWD
GPON_PLOAM_PASSWD=47344436463231314300
# flash get LOID
LOID=123456789
# flash get LOID_PASSWD
LOID_PASSWD=123456
# flash get HW_SERIAL_NO
HW_SERIAL_NO=ZTELQDLP3H05829
# flash get OMCI_SW_VER1
OMCI_SW_VER1=V7.0.10P1N37
# flash get OMCI_SW_VER1
OMCI_SW_VER1=V7.0.10P1N37
# flash get OMCC_VER
OMCC_VER=178
# flash get OMCI_OLT_MODE
OMCI_OLT_MODE=21
# flash get OMCI_VENDOR_PRODUCT_CODE
OMCI_VENDOR_PRODUCT_CODE=15
# flash get OMCI_CUSTOM_ME
OMCI_CUSTOM_ME=65536
# flash get GPON_ONU_MODEL
GPON_ONU_MODEL=F601V7.0
# flash get OMCI_TM_OPT
OMCI_TM_OPT=2
# flash get OMCI_FAKE_OK
OMCI_FAKE_OK=1
# flash get PON_VENDOR_ID
PON_VENDOR_ID=ZTEG
# flash get VLAN_CFG_TYPE
VLAN_CFG_TYPE=0
# flash get VLAN_MANU_MODE
VLAN_MANU_MODE=1
# flash get PON_MODE
PON_MODE=1
# flash get OMCI_VEIP_SLOT_ID
OMCI_VEIP_SLOT_ID=255
# flash get ELAN_MAC_ADDR
ELAN_MAC_ADDR=3ca7ae058ee4
# flash get OUI
OUI=3c,a7,ae
# flash get MAC_KEY
MAC_KEY=ac98087a48724b9ba093396ab408a3fd
```

erdemefe90, 2024-04-29:
@Anime4000 Don't you have any comment?

ToTheCLI, 2024-05-01:
The Xicom SFP is the same hardware as ODI you can use any ODI firmware on it, show the output of mib 84 and 171 by telnet commands:
omcicli mib get 84

omcicli mib get 171

erdemefe90, 2024-05-01:
the outpus is 
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 75
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 1600
FwdOp:  0x10
NumOfEntries: 1
=================================

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x02
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x00
OutputTPID: 0x00
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 35, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1600, TPID 0
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 75, TPID 4
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================
```

ToTheCLI, 2024-05-01:
> INDEX 0
> Filter Outer   : PRI 15,VID 4096, TPID 0
> Filter Inner   : PRI 8,VID 35, TPID 0, EthType 0x00
> Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
> Treatment Inner   : PRI 8,VID 1600, TPID 0

The above index 0 changes VLAN 35 to 1600, try VLAN 1600, mib171 is not followed by Realtek chipset SFPs like ODI, Lantiq will follow mib171 correctly and you can use VLAN 35 like original ONU Like FS SFP.

erdemefe90, 2024-05-01:
how can i change the vlan 1600? from ODI or on mikrotik.

sfp-sfpplus1 interface tagged to vlan35, and vlan35 selected by my pppoe_client interface on mikrotik. Is it correct?

ToTheCLI, 2024-05-01:
> how can i change the vlan 1600? from ODI or on mikrotik.
> 
> sfp-sfpplus1 interface tagged to vlan35, and vlan35 selected by my pppoe_client interface on mikrotik. Is it correct?

Yes, just use VLAN 1600

erdemefe90, 2024-05-01:
Thank you very much :)

Mikrotik gave me the error "we could not verify ourselves for peers" a few times, then it connected by itself

ToTheCLI, 2024-05-01:
> Thank you very much :)
> 
> Mikrotik gave me the error "we could not verify ourselves for peers" a few times, then it connected by itself

Is the Upload and Download the same as Original?

erdemefe90, 2024-05-01:
yes my result is similar with ZTE ONT araund 920-930mbps but SFP is connected as 1Gbps altough LAN_SDS_MODE=4
[image]

erdemefe90, 2024-05-21:
Hello again,

Today my connection lost suddenly. My guess my ISP changed some configs. I saw the connection when I connected the ISP's OLT back  from vlan35. 

Obviously SFP-OLT mib84 query is empty like 
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

ZTE's mib result like this. It is different according to my older result.
```
##################################
MIB INFO:
         ME CLASS: 84
         DB NAME: vlan_filter, DBHandle: 69
##################################

<-----MeID[ 0x01ff,511 ], Addr[ 0x1edfe1]----->
 vlan_filter_list:00 00
                          00 00 00 00 00 00 00 00 00 00
                          00 00 00 00 00 00 00 00 00 00
                          00 00
      forward_opr:04
           number:00

<-----MeID[ 0x0003,3 ], Addr[ 0x243769]----->
 vlan_filter_list:03 03
                          00 00 00 00 00 00 00 00 00 00
                          00 00 00 00 00 00 00 00 00 00
                          00 00
      forward_opr:10
           number:01

<-----MeID[ 0x0004,4 ], Addr[ 0x2457c1]----->
 vlan_filter_list:00 4b
                          00 00 00 00 00 00 00 00 00 00
                          00 00 00 00 00 00 00 00 00 00
                          00 00
      forward_opr:10
           number:01
```

ToTheCLI, 2024-05-21:
> Obviously SFP-OLT mib84 query is empty like
> 
> ```
> # omcicli mib get 84
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> VlanTagFilterData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> # omcicli mib get 171
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> ExtVlanTagOperCfgData
> XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
> ```

Does it still have O5 status?

erdemefe90, 2024-05-21:
yes O5

ToTheCLI, 2024-05-21:
Try VLAN 771 on ODI

erdemefe90, 2024-05-21:
ZTE's MIB84 query data previously showed hex(06 40) dec(1600). Currently it says hex(03 03) dec(771) in the same place. Accordingly, I tried to connect to Mikrotik with vlan 771, but it did not work.

erdemefe90, 2024-05-21:
> Try VLAN 771 on ODI

So, which VLAN should I assign to the sfp-plus1 interface in mikrotik?

ToTheCLI, 2024-05-21:
The most likely explanation is you have fake O5...

ToTheCLI, 2024-05-21:
Maybe try:
flash set OMCI_FAKE_OK 1

erdemefe90, 2024-05-28:
> Maybe try: flash set OMCI_FAKE_OK 1

did not change anything.

I looked at the link you sent. I set all OMCI related things. But still MIB 84 and MIB 171 got empty. I can not understand my O5 state fact or fake. Do you have any suggestion? What can i do? This system worked a few weeks with same router same SFP and same ISS. 
That's why I don't want to give up

ToTheCLI, 2024-05-28:
> did not change anything.
>
> I looked at the link you sent. I set all OMCI related things. But still MIB 84 and MIB 171 got empty. I can not understand my O5 state fact or fake. Do you have any [...]

Well you can try Lantiq based GPON SFPs like this one:
https://www.fs.com/products/133619.html
Also try chaning OMCI OLT MODE try 1 and 21

ToTheCLI, 2024-05-28:
flash set OMCI_OLT_MODE 21
reboot
Check MIB 84 and 171
Then try
flash set OMCI_OLT_MODE 1
reboot

erdemefe90, 2024-05-29:
it is already 21 right now. because when i change it, my software version changes itself then onu state changes between O3 and O5

Anime4000, 2024-05-29:
on your ZTE ONU, can you get MIB 350 until 399 (Vendor Specific OMCI)

erdemefe90, 2024-05-29:
> on your ZTE ONU, can you get MIB 350 until 399 (Vendor Specific OMCI)

There is no data between mib 350 and 399

DanielWONGFS, 2024-09-08:
Erdem, i am having similar issue as what you are having here.
have you resolve this?

may i also ask how are you able to telnet into your ZTE ZXHN F601?
thanks much

erdemefe90, 2024-09-08:
Yes i resolve. I enable telnet while fallowing below link. 
My the last problem the VLAN id was changed. I found the new VLAN id with OMCI ME 171 command and enter the new vlan into SFP GPON my problem was fixed.

https://github.com/douniwan5788/zte_modem_tools

tahaakar-2, 2024-10-06:
Hi everyone how are you 
I have fiberhome router and its bad for gaming 
I try to get stable network by use media converter i have ufiber instant 
So i clone every thing in my fiberhome include rtk infos except vlan which is 126/4 
I have problem o3-o5 loop can any one help my with that please thanks

alexeymx, 2024-12-22:
> Yes i resolve. I enable telnet while fallowing below link. My the last problem the VLAN id was changed. I found the new VLAN id with OMCI ME 171 command and enter the new vlan into SFP GPON my problem was fixed.
> 
> https://github.com/douniwan5788/zte_modem_tools

Which mode (stick or gateway ?) and firmware you use ? I’ve exact same problem with zte and ODI stick as well

cjllin, 2025-03-25:
hello sir,can you tell me how to get ZTE OMCI MIB LOG？can you tell me ZTE olt how to print OMCI/OAM LOG  ？Thank you very much!

erdemefe90, 2025-03-25:
Hello,
I followed the https://hack-gpon.org/ont-zte-f601/ link. You should enable the telnet access  firstly via a python script. Then commands are like these

`redir printf`
`sendcmd 132 omcidebug showmedata 7` (7 is the MIB code for ex. sw version)