Anime4000/RTL960x #283: O5 but no ethernet, omcicli no output

blackshadev, 2024-04-28:
I am struggling to get a UFiber module to work with my ISP.  I have the stick in my EdgeRouter 4, which was "working" with a "GPON-ONU-34-20BI" from FS.com (my ISP allows me to run my own module). On my edgerRouter I filter for the ISPs vlan (100). I have some weird issues with it (sometimes my internet freezes), that is why I am looking at alternatives. I copied the GPON_SN over from the old module and the onu-state goes to O5. But I do not get any internet.

When running
```
diag
> gpon get onu-state 
```

It  says it is on O5. But I still do not have any internet through it. 

To diagnose I ram
```
omcicli mib get 84
```

which doesn't return anyting.

The other diagnosing tool  `diag` does work. I ran it to determine it the onu-state and to output l2-table, which does seem to contain VLAN 100. 
```
diag  l2-table get entry address valid
> l2-table get entry address valid
```
The output I have attached below.

Can anybody shine some light on this issue? Any pointers what to try or diagnose next?

I already tried 
- https://github.com/Anime4000/RTL960x/blob/main/Docs/fakeO5.md
- Setting OMCI_CUSTOM_BDP to 0
- Setting all OMCI flash values to something resembling my fs.com stick (although I don't think it matters)
- Using OMCI_FAKE_OK
- Setting VLAN_CFG_TYPE 1 and VLAN_MANU_MODE 0

```
diag l2-table get entry address valid
l2-table get entry address valid 

LUT address: 0x0000 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
F4:92:BF:8F:3F:69 2   0   6   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x0008 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
74:AC:B9:5C:EF:D0 2   0   7   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x000c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
00:00:00:01:00:02 3   0   1   0    Static 0    SVL
CtagIf Auth DaBlock SaBlock Arp
Dis    Dis  Dis     Dis     Dis

LUT address: 0x0028 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash

--More--command:
D0:21:F9:5A:5B:03 2   0   7   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x003c (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
[... 329 lines cut ...]
En     Dis  Dis     Dis     Dis

LUT address: 0x03d0 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
B0:F2:08:19:3D:9A 2   0   2   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x03e0 (1K LUT)
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
74:AC:B9:5F:23:E5 2   0   5   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis

LUT address: 0x03e8 (1K LUT)

--More--
L2 Unicast table:
MACAddress        Spa Fid Age Vid  State  Ext  Hash
D0:2E:15:00:02:13 2   0   3   100  Auto   0    SVL
CtagIf Auth DaBlock SaBlock Arp
En     Dis  Dis     Dis     Dis
```

Anime4000, 2024-04-28:
can you share output for:
`omcicli mib get 256` and `omcicli mib get 257`
see what OLT see from your UFiber

blackshadev, 2024-04-28:
Thanks for the fast reply

All omcicli commands don't return any output. That is why I am stumped most of your great debugging resources assume these output something, while on my module, they don't.

 
[image]

Anime4000, 2024-04-29:
it appear firmware omitting the output, like this we can troubleshoot

best interest is to purchase ODI Stick instead

blackshadev, 2024-04-29:
That's a shame, but understandable. Do you got a full product name or link to a webshop?

Anime4000, 2024-05-01:
buy here https://www.aliexpress.com/item/1005003515456552.html

closed 2024-05-01 by Anime4000