Anime4000/RTL960x #284: Rx drop

qatar2030, 2024-05-06:
I am using DFP-34X-2C2 from ODI with latest firmware V1.1.8-240408 and I have a problem with Rx drop who to fix this issue please
[image]

lehoangnb, 2024-05-09:
Set egress rate on router to 1244Mbps

Lainbutcher, 2024-05-10:
Hello, There are also drops of RX packets on the LAN port. I tried firmware 220304, 240202, 240407, 240408. On all these firmware there are drops of RX packages. I tested it on Mikrotik CCR1036 equipment and on Cisco WS-C3560CX-12PD-S (with and without optics inserted). That's why the previous advice doesn't work... You need to think about something, dropping packages is bad.

Anime4000, 2024-05-10:
the PON Stick has very small CPU and Memory, RX/TX Flow Control hard for the Stick, larger memory like Huawei MA5671a have support Flow Control.

as @lehoangnb said before, since you have CCR1036, set the limit at Switch Port (if available)

Lainbutcher, 2024-05-10:
> the PON Stick has very small CPU and Memory, RX/TX Flow Control hard for the Stick, larger memory like Huawei MA5671a have support Flow Control.
> 
> this because GPON (2488Mbps/1244Mbps) is not match with 2.5Gb (2500Mbps/2500Mbps), this packet drop is happen due to absent of Flow Control
> 
> as @lehoangnb said before, since you have CCR1036, set the limit at Switch Port (if available)

Thank you so much for the reply. There is no switch chip on the CCR1036, as I understand it, or I have not yet found the settings you are writing about... but it is possible to manually queue packages in the Interface Queues section, and not to the full extent of the hardware, and this I achieved that Tx Queue Drops disappear at Speedtest Uploads. But here's the problem and the question, these errors go to the DFP-34X-2C3 menu on the LAN interface... and I set the speed not 2500, but 1000 on both the DFP-34X and the Microtica interface... They're coming anyway...

P.S.
But the most important thing is that errors occur when installing the DFP-34X in a Cisco switch without optics turned on, at a speed of 1Gbps.

lehoangnb, 2024-05-11:
> Thank you so much for the reply. There is no switch chip on the CCR1036, as I understand it, or I have not yet found the settings you are writing about... but it is possible to manually queue [...]

I've try to set interface queue, but it didn't help too much (you can use wondersharper on linux). If your ISP did not provide over 1Gbps speed, you may change sfp speed to SGMII (1Gbps), it will reduce rx drop

Lainbutcher, 2024-05-11:
Yes, my ISP provider limits the connection speed by 1Gb on its OLT port. But why do errors occur without connected optics and with the set Router LAN interface speed 1Gb?