Anime4000/RTL960x #285: Replacing optical cable equipment without any information.

Kamilake, 2024-05-08:
Hi! First of all, I apologize if I posted this in the wrong place. It's not an issue, it's a simple question.

I am using `SK Broadband` in Korea and there is nothing visible as a MAC address or S/N on the provider's Stock ONU.

However, after checking this repository, it appears that several additional information is required to pass the provider's authentication.
In this situation, is there any way to extract additional information from the device or extract the MAC or serial?

Thank you in advance!

[image]

[image]

vuducdong, 2024-05-08:
@Kamilake 

- GPON SN is DGNT19470042

It appears that this modem's web management credentials label has been removed

I saw someone else similar to this: https://www.clien.net/service/board/kin/17691729

You can rely on the LAN IP when connecting to the modem to find the gateway IP to access the management web, or there may be a separate app for this modem.

Exam: https://m.blog.naver.com/seoyi19/222485272687

Kamilake, 2024-05-10:
> You can rely on the LAN IP when connecting to the modem to find the gateway IP to access the management web, or there may be a separate app for this modem.

Thank you for quick response! Unfortunately it doesn't have NAT and appears to just act like an unmanaged switch. I'm still looking for a solution, but I may have to buy an OLT and find it myself..!

Anime4000, 2024-05-10:
> but I may have to buy an OLT and find it myself..!

this is neat solution and able to read all ME

mog422, 2024-10-14:
사진에 올려주신 DGNT 로 시작하는 12자리가 GPON Serial 입니다.

Kamilake, 2024-10-15:
> 사진에 올려주신 DGNT 로 시작하는 12자리가 GPON Serial 입니다. 저는 FS GPON-ONU-34-20BI로 SKBB 붙는데 성공했습니다.

헉 감사합니다..! 도전해 볼게요!