Anime4000/RTL960x #287: FINALLY SOLVED - ODI DFP-34X-2C2 ON HUAWEI ONLY OLT REGION - MIBs INSERTED ON FIRMWARE BY TRIPLE OXYGEN.

techtito, 2024-05-10:
Hello guys, i am Tito from Brazil.
I am trying until 2 months to make the ODI DFP-34X-2C2 work on my ISP who use only HUAWEI HGU and OLT in my city WITHOUT SUCCESS.

But there is an great Brazilian programmer called TRIPLE OXYGEN who made an MODIFIED firmware for VSOL V2802RH the version is 1.9.4 he have INSERTED MIBs and MEs for HUAWEI specific OLT because there is another guy in Brazil who had SAME problem of me and WORKED WELL.

So i have send an message via e-mail for Triple Oxygen and after i have explained my case he prepared an modified firmware with MIBs of Huawei OLT for me, i have send for him the LATEST firmware was published o Realtek Chipset Discord Server, its:
M110_sfp_ODI_240202
He have modified and inserted the MIBs and changed name to:
V1.1.6-240203
the file is:
https://tripleoxygen.net/files/devices/odi/dfp-34x-2c2/firmware/custom/V1.1.6-240203.tar

After uploading the firmware the only thing i need to do to get fully sync and REAL O5 state is:
flash set GPON_SN HWTCxxxxxxxx - YOUR GPON SN
flash set OMCC_VER 136 - THIS IS MANDATORY - ON VSOL ITS: "cfgmib set OMCC_VER 0x86" BUT ON ODI IS 136.
flash set PON_VENDOR_ID HWTC
flash set OMCI_OLT_MODE 1
flash set OMCI_TM_OPT 2
reboot

This new firmware have a lot of new and cool features like:
Show OLT manufacturer on the web gui
Show VLAN when you are FULLY AUTENTICATED and NOT FAKE O5 i was.
There are a lot of backup profiles from various HGU used around world.

I will put some new prints here of my final result on ODI, i have gained some download and upload speed, my ping was before 41-42ms pushed down for 37-38ms for same server.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

THANKS FOR TRIPLE OXYGEN IN BRAZIL
i hope i can help ANYONE who have the SAME problem with HUAWEI OLT.

Anime4000, 2024-05-10:
# the file is removed, perhaps you can upload here for Brazil ISP

Version after > V1.1.0 has all these by default, I also working on improving to mitigate fake O5, as well newer WebGUI and build-in MAC Keygen

**YouTube Video:**
[image]

**Screenshot, based on V1.1.0 220923, W.I.P.**
[image]
[image]

techtito, 2024-05-10:
you can download and analyse the firmware by using the mib get commands on telnet.

Anime4000, 2024-05-10:
Upon checking, my ISP didn't use ME 350, 370 and 373 (Reserved for vendor-specific use), it show blank Entity. as these ME is not part of Universal ONU as my country standards 😅

diogosilvan, 2024-05-10:
> # the file is removed, perhaps you can upload here for Brazil ISP
> Version after > V1.1.0 has all these by default, I also working on improving to mitigate fake O5, as well newer WebGUI and build-in MAC Keygen
> 
> **YouTube Video:** [image]
> 
> **Screenshot, based on V1.1.0 220923, W.I.P.** ![04](https://private-user-images.githubusercontent.com/1908715/329678435-5977b9df-2154-4bec-a170-0da44d5350c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzNzI4OTQsIm5iZiI6MTcxNTM3MjU5NCwicGF0aCI6Ii8xOTA4NzE1LzMyOTY3ODQzNS01OTc3YjlkZi0yMTU0LTRiZWMtYTE3MC0wZGE0NGQ1MzUwYzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTBUMjAyMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmE1MWFkY2E5MDdhY2RiOTdkMGI3MDA5MzhiZTNkZjJiOWFmODk0YTYwNzFjYTQwMzFlNjg5NDNjMDViOWVjNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Y1sd83swIlrhZzsW1FkCnARwsqdtvEZBAnPc3EvlqBg) ![06](https://private-user-images.githubusercontent.com/1908715/329678451-ec0630f2-3d0d-4d15-9929-2e827f6037ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzNzI4OTQsIm5iZiI6MTcxNTM3MjU5NCwicGF0aCI6Ii8xOTA4NzE1LzMyOTY3ODQ1MS1lYzA2MzBmMi0zZDBkLTRkMTUtOTkyOS0yZTgyN2Y2MDM3ZWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTBUMjAyMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWRhZjEwYzY3NGM1MjQ5ZDIwZWIxZDgxMjhhYjc0NDc3YjExOGQyYjM4NDkwNTNmMzQ5NDlkZjNhYjIwZjIzMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jXUwQYleo1v_otZNhcsK4SLoSIagqXZL2MIQpJXFnZI)

How can I get this mod?

techtito, 2024-05-10:
> How can I get this mod?

Here:
https://tripleoxygen.net/files/devices/odi/dfp-34x-2c2/firmware/custom/V1.9.1-240202.tar

techtito, 2024-05-10:
> Upon checking, my ISP didn't use ME 350, 370 and 373 (Reserved for vendor-specific use), it show blank Entity. as these ME is not part of Universal ONU as my country standards 😅

This is complicated because i really believe it will be appear more people around the world with this same problem.
If its possible for you, please insert these ME on next firmwares for ODI, here in Brazil a lot of people who wants to take off the ISP HGU already know the triple oxygen, but he only did it before i asked for him to do it, in the end its great because we already have all options, but if you get know about someone same problem, remember that can be specific ME on Huawei OLT, in my hole country only huawei OLT have this behavior.
the old dont care for mac address/mac key, and a lot of other things, the OLT just want GPON SN + specific ME for this case.

diogosilvan, 2024-05-10:
> <https://private-user-images.githubusercontent.com/1908715/329678435-5977b9df-2154-4bec-a170-0da44d5350c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzNzI4OTQsIm5iZiI6MTcxNTM3MjU5NCwicGF0aCI6Ii8xOTA4NzE1LzMyOTY3ODQzNS01OTc3YjlkZi0yMTU0LTRiZWMtYTE3MC0wZGE0NGQ1MzUwYzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTBUMjAyMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmE1MWFkY2E5MDdhY2RiOTdkMGI3MDA5MzhiZTNkZjJiOWFmODk0YTYwNzFjYTQwMzFlNjg5NDNjMDViOWVjNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Y1sd83swIlrhZzsW1FkCnARwsqdtvEZBAnPc3EvlqBg> [image:

Does this version have the modified webgui that you sent the prints?

techtito, 2024-05-10:
> […](#)

Yes, same I am using now and prints above

Anime4000, 2024-05-11:
Upon check the binary file, 99% file are different (as different hashes) this look like they have Realtek SDK to compile it.

however, I can use my own method to inject. 

can you paste Huawei ME `omcicli mib get ` 350, 370 and 373

from my end with Triple Oxygen Modded firmware, it came up empty because I was under Nokia Universal OLT.

so, your ME value needed

techtito, 2024-05-11:
@Anime4000 thank you very much, sure i can:

# omcicli mib get 350
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
HW ME350
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0000
First: 0x03020000
Second: 0x0000
Third: 0x0200
4: 0x0258
5: 0x0100
6: 0x0000
7: 0x00
8: 0x00000000
9: 0x000000
10: 0x00
11: 0x00
12: 0x00
13: 0x00
14: 0x01080108
15: 0x053977053977
16: 0x0000
=================================
# omcicli mib get 370
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
HW ME370
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0101
First: 0x0201
Second: 0x01000000
Third: 0x00000000
=================================
# omcicli mib get 373
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
HW ME373
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0000
NumOfParts: 3
Part1: EchoLife HG8010H GPON Ter
Part2: minal (CLASS B+/PRODUCT I
Part3: 0x443a3231303233313142515036524732303030303038290000
Part4: 0x00000000000000000000000000000000000000000000000000
Part5: 0x00000000000000000000000000000000000000000000000000
Part6: 0x00000000000000000000000000000000000000000000000000
Part7: 0x00000000000000000000000000000000000000000000000000
Part8: 0x00000000000000000000000000000000000000000000000000
Part9: 0x00000000000000000000000000000000000000000000000000
Part10: 0x00000000000000000000000000000000000000000000000000
Part11: 0x00000000000000000000000000000000000000000000000000
Part12: 0x00000000000000000000000000000000000000000000000000
Part13: 0x00000000000000000000000000000000000000000000000000
Part14: 0x00000000000000000000000000000000000000000000000000
Part15: 0x00000000000000000000000000000000000000000000000000
=================================

Can you please tell me what do you think about it?

And o also paste below the omcicli mib get all for checking:

# omcicli mib get all
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Anig
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x8001
SRInd: 1
[... 5406 lines cut ...]
=================================
EntityId: 0x0601
AdminState: 0
OperState: 0
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagOpCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
HSQDefault
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
WANConfigCata
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
WANConfigCataDeti
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
#

Anime4000, 2024-05-11:
at least for MA5671a PON Stick, can edit MIB file and add new entry like this:
`/etc/mibs/data_1g_8q_us1280_ds512.ini`
```
# Huawei ME (Brazil)
350 0x0000 0x03020000 0x0000 0x0200 0x0258 0x0100 0x0000 0x00 0x00000000 0x000000 0x00 0x00 0x00 0x00 0x01080108 0x053977053977 0x0000
370 0x0101 0x0201 0x01000000 0x00000000
373 0x0000 3 "EchoLife HG8010H GPON Ter" "minal (CLASS B+/PRODUCT I" 0x443a3231303233313142515036524732303030303038290000 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
```

techtito, 2024-05-11:
> at least for MA5671a PON Stick, can edit MIB file and add new entry like this: `/etc/mibs/data_1g_8q_us1280_ds512.ini`
>
> ```
> # Huawei ME (Brazil)
> 350 0x0000 0x03020000 0x0000 0x0200 0x0258 0x0100 [...]

after this great work please let me know how versions of firmware for ODI will be modified, thank you.

Anime4000, 2024-05-11:
I can try to work on it, but I not sure how it can be done as I no way to verify,
however, I have send these to ODI Developer if ODI can make it permanent on future veersion

Anime4000, 2024-05-11:
@techtito 
Much easier is let ODI did it, I have send your ME 350, 370, 373 to ODI Devs via whatsapp, hope ODI make into future version.

Anime4000, 2024-05-11:
@diogosilvan if you like to try new GUI, can get it here:
https://github.com/Anime4000/RTL960x/discussions/286#discussioncomment-9395068

techtito, 2024-05-11:
> I can try to work on it, but I not sure how it can be done as I no way to verify, however, I have send these to ODI Developer if ODI can make it permanent on future veersion

If you want i cant test the firmware for you, and its Nice to hear about you sending for ODI developer also, i hope they include

techtito, 2024-05-12:
Hello @Anime4000 @tripleoxygen was send for me this page with instructions to insert the MEs from huawei here in github, please take a look:
https://github.com/tripleoxygen/realtek-libohwtc

Anime4000, 2024-05-12:
@techtito I read the @tripleoxygen [realtek-libohwtc](https://github.com/tripleoxygen/realtek-libohwtc)
I don't have `uClibc MIPS/Lexra GCC toolchain`, the readme said simple as copy to feature folder, so I forward to V1.1.8 240408 
[M110_sfp_ODI_240408_rel240513+hwtc_git.zip](https://github.com/Anime4000/RTL960x/files/15287006/M110_sfp_ODI_240408_rel240513%2Bhwtc_git.zip)
unzip and flash the firmware, hope it works,

if not, switch back to working partition by:
```
nv getenv sw_active
```
If return 0, set 1
If return 1, set 0
```
nv setenv sw_active X
nv setenv sw_commit X
```

techtito, 2024-05-12:
Hello @Anime4000 i have tested right now and it also works very well, this is great!
some pics below:
SFP:
[image]

PON:
[image]

OLT:
[image]

Mib GET 350/370/373:
[image]

VLAN:
[image]

Statistics:
[image]

My ISP OLT only wants the GPON SN + OMCC + OLT MODE and correct mibs for work, OLT dont care for mac address, SN and other things.

lehoangnb, 2024-05-13:
> @techtito I read the @tripleoxygen [realtek-libohwtc](https://github.com/tripleoxygen/realtek-libohwtc) I don't have `uClibc MIPS/Lexra GCC toolchain`, the readme said simple as copy to feature folder, so I forward to V1.1.8 240408
> 
> [M110_sfp_ODI_240408_rel240513+hwtc-fix.zip](https://github.com/Anime4000/RTL960x/files/15287033/M110_sfp_ODI_240408_rel240513%2Bhwtc-fix.zip)
> 
> unzip and flash the firmware, hope it works, also read `flash set for huawei.txt` there are two extra command in case SW VER not working
> 
> if not, switch back to working partition by:
> 
> ```
> nv getenv sw_active
> ```
> 
> If return 0, set 1 If return 1, set 0
> 
> ```
> nv setenv sw_active X
> nv setenv sw_commit X
> reboot
> ```

I have compiled mips gcc if you interest [mips-buildroot-linux-uclibc-gcc](https://github.com/lehoangnb/mips-buildroot-linux-uclibc-gcc)

Anime4000, 2024-05-13:
@lehoangnb this can be run on Realtek Lexra?
can you tell us how to use it?

lehoangnb, 2024-05-13:
> @lehoangnb this can be run on Realtek Lexra? can you tell us how to use it?

Its can be used for compile realtek-libohwtc, just export patch to bin folder and run make command in realtek-libohwtc directory
You can use this gcc to compile binary for realtek mips chip, its based uclibc (have prebuilt in SFP firmware)

Anime4000, 2024-05-15:
@techtito I notice OLT Page is broken, here the fixed version, of course with Huawei MIB from tripleoxygen
[M110_sfp_ODI_220923-240408_rel240515+mib.zip](https://github.com/Anime4000/RTL960x/files/15316413/M110_sfp_ODI_220923-240408_rel240515%2Bmib.zip) contain 220923 (VEIP Support) and 240408 (VEIP/ALCL)

overall, how do you like the new GUI?

closed 2024-05-15 by Anime4000

diogosilvan, 2024-05-15:
> @techtito I notice OLT Page is broken, here the fixed version, of course with Huawei MIB from tripleoxygen [...]

Does this have the new GUI? I thought the new GUI was very good.

Anime4000, 2024-05-15:
Yes, this new GUI, still in beta stage, so, I still finding the bug

vuducdong, 2024-05-15:
@techtito 

Has anyone used it with iptv multicast?

How can I contact TRIPLE OXYGEN?

diogosilvan, 2024-05-15:
> Yes, this new GUI, still in beta stage, so, I still finding the bug

 For me, I found a possible bug.  IGMP Snooping seems to be exchanged, the option to enable this by disabling. 

 In addition, VLAN Forward Operation is not working here. I will do more tests to confirm 100%.

diogosilvan, 2024-05-15:
> @techtito
> 
> Has anyone used it with iptv multicast?
> 
> How can I contact TRIPLE OXYGEN?

 Here with the multicast with OLT Huawei remains broken. If the multicast is active the SFP stops responding to all traffic until the multicast traffic stops.

Anime4000, 2024-05-15:
> In addition, VLAN Forward Operation is not working here. I will do more tests to confirm 100%.

which format are you enter `Id:FwdOp`? I notice FwdOp hack work on `220923`

lehoangnb, 2024-05-15:
> > In addition, VLAN Forward Operation is not working here. I will do more tests to confirm 100%.
> 
> which format are you enter `Id:FwdOp`? I notice FwdOp hack work on `220923`

Change cut command in your fix script to awk because 240408 removed cut command in busybox

Anime4000, 2024-05-15:
> Change cut command in your fix script to awk because 240408 removed cut command in busybox

crap, I didn't notice that

myepezg, 2024-05-15:
Hello @techtito I tried with this version https://tripleoxygen.net/files/devices/odi/dfp-34x-2c2/firmware/custom/V1.9.1-240202.tar

But, I had 10Mbps upload and 300Mbps download. And thats not my real speed.

lehoangnb, 2024-05-16:
> > > @techtito
> > > Has anyone used it with iptv multicast?
> > > How can I contact TRIPLE OXYGEN?
> > 
> > 
> > Here with the multicast with OLT Huawei remains broken. If the multicast is active the SFP stops responding to all traffic until the multicast traffic stops.
> 
> yes, that's correct.

I'm facing problem with upload, it's really slow if upload to international server (like from Vietnam to Singapore), does this problem is related to multicast?

diogosilvan, 2024-05-16:
> > In addition, VLAN Forward Operation is not working here. I will do more tests to confirm 100%.
> 
> which format are you enter `Id:FwdOp`? I notice FwdOp hack work on `220923`

I tryed both versions. In 220923 really works VLAN Forward Operation (they change FwdOp as well) but no make differece in pratice.
I'm use on fix 0xe103:0x02. Use as base the command "omcicli mib set 84 0xe103 FwdOp 0x02" because works very well if set in telnet/ssh or script in startup.

And if in two versions i really dont know way, if i set manualy in terminal "omcicli mib set 84 0xe103 FwdOp 0x02" still not working. But if i flash previous version "M110_sfp_ODI_220817.tar" only with a modded script to fix vlan by me to works very well but have only old GUI.

I checked all configs a lot times, in terminal and GUI everyting are ok but VLAN don't works. And i will try debug, if had any update i send here.

I'm use in Oi (VTAL) Brazil, Nokia ALCL on both sides.

diogosilvan, 2024-05-16:
> I'm facing problem with upload, it's really slow if upload to international server (like from Vietnam to Singapore), does this problem is related to multicast?

If you try to a local server, upload speed are ok or the same?

For me multicast stop all traffic form SFP. You use for multicast with sucess in your case?

lehoangnb, 2024-05-17:
> If you try to a local server, upload speed are ok or the same?
>
> For me multicast stop all traffic form SFP. You use for multicast with sucess in your case?

Local speed are fine (~2.4Gbps up/down), I removed multicast in downstream (diag gpon del ds-flow flow-id ...) but still problem