Anime4000/RTL960x #288: HG322G xpon with Realtek RTL9601D

aryanrajseo, 2024-05-11:
This is not discussed anywhere but HG322G xpon have Realtek RTL9601D chip however the telnet is different. How can we find the available commands and latest firmware? telnet enabled via ui seems to get deactivated after logged out.

Product https://www.vsolcn.com/wp-content/uploads/2023/03/XPON-HGU-HG322G-Datasheet-V2.0-EN.pdf

available in India as

Netlink HG322G https://drive.google.com/file/d/1Usx0bxN9E4fVT937EnHmYvdaSNd88phk/view
Syrotech sy-gpon-1010-dont (confirmed and attached)
DBC HGU-V110
Digisol DG-GR6011

can't confirm the chip in other brand except syrotech.

[image]

Anime4000, 2024-05-11:
this only 1GbE? also notice that it have VoIP

aryanrajseo, 2024-05-11:
It is 1GE. and Yes it have VoIP.

varkey, 2024-07-27:
@aryanrajseo Did you figure out a way to change the GPON serial for this device? I have the Digisol variant.

varkey, 2024-07-27:
So once you enable SSH from the UI (Security --> Login Privilege), you would able to login to a stripped down shell. Once in this shell enter the command `enterlinuxshell` to enter the proper shell. 

```
(⎈)➜  ~ ssh admin@192.168.3.1
admin@192.168.3.1's password:
>enterlinuxshell
#
```

Once in the linux shell, you can follow the same steps as the [VSOL V2801F](https://hack-gpon.org/ont-vsol-v2801f/) to change the GPON serial and other parameters.

qatar2030, 2024-11-15:
> This is not discussed anywhere but HG322G xpon have Realtek RTL9601D chip however the telnet is different. How can we find the available commands and latest firmware? telnet enabled via ui seems to get deactivated after logged out.
> 
> Product https://www.vsolcn.com/wp-content/uploads/2023/03/XPON-HGU-HG322G-Datasheet-V2.0-EN.pdf
> 
> available in India as
> 
> Netlink HG322G https://drive.google.com/file/d/1Usx0bxN9E4fVT937EnHmYvdaSNd88phk/view Syrotech sy-gpon-1010-dont (confirmed and attached) DBC HGU-V110 Digisol DG-GR6011
> 
> can't confirm the chip in other brand except syrotech.
> 
> ![WhatsApp Image 2024-05-11 at 9 36 17 AM](https://private-user-images.githubusercontent.com/34836534/329752288-c646bfec-8586-4eb7-a005-7c996165bfd3.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE2NjEwNzAsIm5iZiI6MTczMTY2MDc3MCwicGF0aCI6Ii8zNDgzNjUzNC8zMjk3NTIyODgtYzY0NmJmZWMtODU4Ni00ZWI3LWEwMDUtN2M5OTYxNjViZmQzLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExNVQwODUyNDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YTI0MWIzZjgyNzBkYjIxYTdkZDkxY2ZiMWU0Y2UxZGI0NDcwZTIxN2VjZGMwY2NmZjc2YjAyNmEzNmNmZWMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.8M7uGwPI8Wr3wDPVRXtMan8ScgZHW6xDCsflpW4UmnE)

can i change GPON SN, Vendor ID and HW version? i need voip

varkey, 2024-11-15:
@qatar2030 Yes, please check my previous comment for details. I was able to get it work on Airtel FTTH.

aryanrajseo, 2024-11-15:
@varkey could you confirm if we can hav vlan scan working with this?

varkey, 2024-11-15:
@aryanrajseo There's no VLAN scanning option in the UI. Although, you wouldn't need it, Airtel's standard VLAN 100 works without any changes. 

You could also lookup the OMCI message from the shell for the available VLANs.

aryanrajseo, 2024-11-15:
@varkey hi, vlan 100 is not standard and changes with router. i have a different vlan. could you help with OMCI message to get the vlan? do you have a video screencast or screenshot?

varkey, 2024-11-18:
@aryanrajseo Actually it is standard for Airtel. However this is the customer side VLAN which remains fixed always, but the internal VLAN would be different. What happens is, when the ONT doesn't do the internal-external VLAN mapping correctly, we would need to find the internal VLAN separately. Anyway with this ONT, the mapping works, so you can directly use VLAN 100 with Airtel. 

I used the `omcicli mib get 171` command and got the below details. As can be seen below, VLAN `100` is the customer side VLAN, which is internally mapped to `1345`. This internal VLAN would differ in each area, however if things are working as expected, we should be able to use VLAN 100 always. 

```
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x09
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x00
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 100, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 1345, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 660, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 666, TPID 4
AssociatedMePoint: 0x601
```

aryanrajseo, 2024-11-18:
Thank you @varkey, that is really helpful. i am currently using XPON HG8120C with custom vlan because i need 2 ports, one for a different location for PTP device. this xpon is useful if you  know vlan id.

https://multicoreinfotech.com/product/fiber-router-ont/xpon-onu-with-voice-ho/

varkey, 2024-11-18:
Even I have the HG8120C (although mine is Huawei branded), excellent device. Even on this unit, I just run it with VLAN 100 and works great.

qatar2030, 2024-11-18:
> @varkey could you confirm if we can have vlan scan working with this?

so how to creat 2 vlan?
vlan10 for internet
vland20 for voip
with regards

aryanrajseo, 2024-11-18:
@qatar2030 not using voip. single vlan id needed for both ports to work. one is 100mbps, 1gbps.

qatar2030, 2024-12-04:
where i can buy it please

qatar2030, 2024-12-11:
sorry i couldn't find it in holl internet

abhijit3691, 2025-03-03:
Hi @varkey , any chance you have the firmware for Digisol DG-GR6011 HW Ver 1.3 mate?
Would appreciate any help, as the latest fw blocks couple of features.

varkey, 2025-03-03:
@abhijit3691 I am on the stock firmware. I don't have the firmware file though. Which version are you on?

| Device model | DG-GR6011 |
| --- | --- |
| Hardware Version | V1.3 |
| Firmware Version | V1.0.00-231212 |

abhijit3691, 2025-03-03:
@varkey I upgraded to V1.1.00-250121.
Do not upgrade, it's all buggy, and they removed Telnet/SSH and the config file is all encrypted or smth.
Idk how to revert back to stock firmware.
Can downgrading brick the device?
I was also curious, since this is VSOL HG322G, can VSOL firmware work on these?

nutp8, 2025-08-14:
@abhijit3691 awesome dude

nutp8, 2025-08-25:
@rajputonline  yeah, telnet is working fine. but ssh is disabled  and the xml config files on  `/etc/config/`    is fully encrypted on the latest firmwares. 

V1.0.00-231212 is the last firmware with readable config files.

nutp8, 2025-08-25:
@abhijit3691 i got the vsol firmware HG322G. i couldn't test it though. 
i tried flashing it. but dg-gr6011 didn't accept another vendor's firmware, got a vendor not matching error.

vsol support is awesome they are fast

file :  https://pixeldrain.com/u/MDfaKa24

SM-03, 2025-09-07:
My old EPON 1GE C ONU got faulty recently, frequent net disconnection issue, used it for around 15 years, so now bought this DIGISOL DG-GR6011 today and noticed on the package sticker the H/W ver is v1.3 and MFD & packing date: May-24

Checked Digisol portal (https://www.digisol.com/firmware) & noticed v1.1. from 2025 so searched the internet for this & redirected to this thread. I'm pretty much a newbie here, and don't know anything much about all these so asking

1. I do I check the current FW version? I'm using Windows. Is it only possible to check on Windows OS, or Linux required?
2. I am using Siti Broadband connection, with TPLink Ax50 Wi-Fi 6 router + this DIGISOL DG-GR6011 now. The Internet is working fine, do I have to make any kind of modification/changes like this FW update?

hridaysarkar, 2026-09-04:
https://drive.google.com/file/d/17OYrfkooIGhYj1MtVv1qB_o4Jpbj2mAI/view?usp=sharing

New DIGISOL DG-6011 Firmware Go Check Out Guys ,It's Now Accessible For Everyone