Anime4000/RTL960x #289: device support

mnewiraq, 2024-05-19:
can you please tell me if the device in the attached photos is supported by your firmwares?
[image]
[image]

Anime4000, 2024-05-19:
I not sure, you need take apart and look the chip inside

mnewiraq, 2024-05-19:
> I not sure, you need take apart and look the chip inside

Would you please inspect the attached photos 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Anime4000, 2024-05-19:
This use very different Processor, also will not work firmware here, as mostly for PON Stick