Anime4000/RTL960x #290: XPONSTICK - How to identity the model without tear down ?

canope-25, 2024-05-20:
Hi guys ,

Hope you're all well.

I received a XPON STICK SFP two days ago.

Look like an ODI one but not sure .

The following two pictures provide a vision of the SFP.
[image]

[image]

[image]

[image]

/proc/meminfo disturb me because it's report 32mbytes of RAM.

Thank you in advance for your help.

Regards,

C...

canope-25, 2024-05-20:
Seems to be the same as mentioned in the following PDF : 

https://github.com/Strykar/GPON/blob/main/PON%20stick%20Spec.pdf

Thank you in advance for your help

Anime4000, 2024-05-21:
it's same, using RTL9601D SoC

closed 2024-05-21 by Anime4000

canope-25, 2024-05-21:
> it's same, using RTL9601D SoC

Thank you , so can use the same firmwares and supporte 2500Base-X ?

Thank you again

Anime4000, 2024-05-21:
it support 2500Base-X natively

canope-25, 2024-05-21:
> it support 2500Base-X natively 

Thank you 🙏.

I'm almost at the end of the tunnel.

The L3 10G switch detected it.
Checking the kernel logs ( through /proc/kmsg ) showed that the sfp stopped at HISGMII PHY).

Having O5 but still not everything as i would like .

My initial ONT is an Alcatel Lucent one ( since 2015 ) without telnet / web access due to the massive vuln identified back in 2016 and for which Orange ( France )decided to shut the web/ telnet interface as soon as the ONT is registered and received an OMCI profil.

My issue know is the fact that I cannot obtain an IPV4 address and the IPv6 over DHCP ( over vlan 832 ).

Will try the legacy PPPOE interface ( over vlan 835 ).

Something is odd, I cloned the configuration at switch level in two port : 
- 1 with a 1G Ethernet Base-T SFP on which the legacy ALU ONT is plugged
- 1 with the ONT SFP 

So at switch level normally everything the same.

Maybe a CoS issue ( but again i rely on patched DHCPv4/v6 for the remarking to COS6 )

I also activated IGMP snooping at SFP level , I received ( without it ) the IPTV multicast flow of the other GPON tree users ( over  vlan 840 )

I would appreciate some help regarding the last issue .

Thank you in advance

canope-25, 2024-05-21:
Hi again , seems really strange .. only the internet vlan have issues , maybe I can share information about that ( can be a wrong mapping at gpon level , it's disturbing )

canope-25, 2024-05-21:
Found out ... seems to be a defect / issue with the stick ...
My ISP expose a dedicated vlan for the internet part ( 832 ) but the stick seems to be unable to remap it correctly and export the technical id part 2800 as vlan in place of 832.

Rely now on a vlan-translation at L3 switch level ... a bit crappy .

Do you know if it's possible to do that through diag / omcicli / etc commands ?

Thank you in advance 🙏

canope-25, 2024-05-22:
> That normal for most Realtek PON Sticks, need use PON Side of VLAN

Hi , thank you for the answer .
So nothing is possible through some diag or ocmcicli command in order to keep the correct mapping ?

My idea is to be able to keep the ONT SFP IN parallel of the legacy GPON ONU and to rely on a splitter + zigbee power management in order to switch back to the legacy ONT in case of internet issue : 
[image]