Anime4000/RTL960x #296: invalid mtu

qatar2030, 2024-05-27:
[image]
[image]
[image]
[image]

What is the causing for that?

lehoangnb, 2024-05-28:
Set pppoe-out mtu to 1492 or 1480, 1460 maybe wrong value for pppoe

Anime4000, 2024-05-28:
can ignore:
https://forum.mikrotik.com/viewtopic.php?t=206991&sid=1c1a5bc0e9d8d63c88561eff8cb764d5

qatar2030, 2024-05-28:
i'm still sufring

ToTheCLI, 2024-05-28:
> i'm still sufring

Did you disable IPv6?

longthanhtran, 2024-06-02:
Looks like your pppoe-client caps at 1460 but you can try with standard 1492 to see if the actual MTU is accepted. There cases where ISPs support [baby jumbo frame](https://www.rfc-editor.org/rfc/rfc4638.html) then you need to set your sfp1plus interface with MTU of 1508 and set your pppoe-client with 1500 to enjoy full MTU 1500.

Anime4000, 2024-06-03:
I did the baby jumbo 1508, somehow ISP use VRRP for PPPoE and lose ability to have 1500 PPPoE MTU, but now only MTU is 1500 is Down Stream path

[image]

longthanhtran, 2024-06-03:
When I look [routeros pppoe client help](https://help.mikrotik.com/docs/display/ROS/PPPoE#PPPoE-PPPoEClient), it apprears that when you don't specify the MTU value, its default value is 1460.

The log entries that show the warning is when a client from your `internal` network try to send packets with MTU 1492 which is larger than gateway MTU then you have fragmentation issue.

Try to use a default MTU 1492 at least to see if it resolves the issue first.

One more thing, as you use VLAN interface in routeros while the sfp can tag VLAN for you, try the option and let your pppoe-client dials on sfp1plus interface.