Anime4000/RTL960x #299: O5 Initial status, but no ONU ID and no ethernet.

xMikooo, 2024-06-03:
Hey I have Leox LXT-010H-D. After cloning GPON SN, SW VER etc I got: 
[image]
Although it has O5 so it's authenticated, internet doesn't work and PON stats are empty
[image]
I would imagine that at least something will be sent. What's also weird is that the fiber interface is DOWN
[image]
Original ONT reports that the OLT is Dasan V8102
[image]

Here's omcilog.par
```
# cat omcilog.par
0x0000 EthUni Alarm(0x0101, NO.00-07=0x00, NO.08-15=0x00)
0x0000 EthUni Alarm(0x0101, NO.00-07=0x80, NO.08-15=0x00)
0x0001 OntData Get(0x0000,MIBDataSync)
0x0001 OntData GetRsp(0x0000,Result=0x0,MIBDataSync=0x00)
0x0000 Anig Alarm(0x8001, NO.00-07=0x08, NO.08-15=0x00)
0x0002 OntData Get(0x0000,MIBDataSync)
0x0002 OntData GetRsp(0x0000,Result=0x0,MIBDataSync=0x00)
0x0003 Anig Get(0x8001,SRInd,NumOfTcont,GemBlkLen,PbDbaRpt,OnuDbaRpt,SFThreshold,SDThreshold,ARC,ARCInterval,OpticalSignalLevel,LowOpThreshold,UppOpThreshold,OntRspTime,TranOpticLevel,LowTranPowThreshold,UppTranPowThreshold)
0x0003 Anig GetRsp(0x8001,Result=0x0,SRInd=0x01,NumOfTcont=0x0010,GemBlkLen=0x0030,PbDbaRpt=0x00,OnuDbaRpt=0x00,SFThreshold=0x05,SDThreshold=0x09,ARC=0x00,ARCInterval=0x00,OpticalSignalLevel=0xd1c8,LowOpThreshold=0xff,UppOpThreshold=0xff,OntRspTime=0x88b8,TranOpticLevel=0x0000,LowTranPowThreshold=0x81,UppTranPowThreshold=0x81)
0x0004 UnknowME 65304
0x0004 UnknowME 65304
0x0000 Anig Alarm(0x8001, NO.00-07=0x00, NO.08-15=0x00)
0x0005 Anig Get(0x8001,SRInd,NumOfTcont,GemBlkLen,PbDbaRpt,OnuDbaRpt,SFThreshold,SDThreshold,ARC,ARCInterval,OpticalSignalLevel,LowOpThreshold,UppOpThreshold,OntRspTime,TranOpticLevel,LowTranPowThreshold,UppTranPowThreshold)
0x0005 Anig GetRsp(0x8001,Result=0x0,SRInd=0x01,NumOfTcont=0x0010,GemBlkLen=0x0030,PbDbaRpt=0x00,OnuDbaRpt=0x00,SFThreshold=0x05,SDThreshold=0x09,ARC=0x00,ARCInterval=0x00,OpticalSignalLevel=0xd1c8,LowOpThreshold=0xff,UppOpThreshold=0xff,OntRspTime=0x88b8,TranOpticLevel=0x0000,LowTranPowThreshold=0x81,UppTranPowThreshold=0x81)
0x0000 EthUni Alarm(0x0101, NO.00-07=0x00, NO.08-15=0x00)
0x0000 EthUni AVC(0x0101, NO.00-07=0x04, NO.08-15=0x00)
0x0000 EthUni AVC(0x0101, NO.00-07=0x02, NO.08-15=0x00)
0x0006 EthUni Get(0x0101,ExpectedType,SensedType,AutoDectectCfg,EthLoopbackCfg,AdminState,OpState,DuplexInd,MaxFrameSize,DTEorDCEInd,PauseTime,BridgedorIPInd,ARC,ARCInterval,PppoeFilter,PowerControl)
0x0006 EthUni GetRsp(0x0101,Result=0x0,ExpectedType=0x2f,SensedType=0x2f,AutoDectectCfg=0x00,EthLoopbackCfg=0x00,AdminState=0x00,OpState=0x00,DuplexInd=0x03,MaxFrameSize=0x07d0,DTEorDCEInd=0x00,PauseTime=0x0000,BridgedorIPInd=0x02,ARC=0x00,ARCInterval=0x00,PppoeFilter=0x00,PowerControl=0x00)
0x0000 Anig Alarm(0x8001, NO.00-07=0x10, NO.08-15=0x00)
0x0007 Anig Get(0x8001,SRInd,NumOfTcont,GemBlkLen,PbDbaRpt,OnuDbaRpt,SFThreshold,SDThreshold,ARC,ARCInterval,OpticalSignalLevel,LowOpThreshold,UppOpThreshold,OntRspTime,TranOpticLevel,LowTranPowThreshold,UppTranPowThreshold)
0x0007 Anig GetRsp(0x8001,Result=0x0,SRInd=0x01,NumOfTcont=0x0010,GemBlkLen=0x0030,PbDbaRpt=0x00,OnuDbaRpt=0x00,SFThreshold=0x05,SDThreshold=0x09,ARC=0x00,ARCInterval=0x00,OpticalSignalLevel=0xd1c8,LowOpThreshold=0xff,UppOpThreshold=0xff,OntRspTime=0x88b8,TranOpticLevel=0x0343,LowTranPowThreshold=0x81,UppTranPowThreshold=0x81)
0x0000 Anig Alarm(0x8001, NO.00-07=0x00, NO.08-15=0x00)
0x0008 Anig Get(0x8001,SRInd,NumOfTcont,GemBlkLen,PbDbaRpt,OnuDbaRpt,SFThreshold,SDThreshold,ARC,ARCInterval,OpticalSignalLevel,LowOpThreshold,UppOpThreshold,OntRspTime,TranOpticLevel,LowTranPowThreshold,UppTranPowThreshold)
0x0008 Anig GetRsp(0x8001,Result=0x0,SRInd=0x01,NumOfTcont=0x0010,GemBlkLen=0x0030,PbDbaRpt=0x00,OnuDbaRpt=0x00,SFThreshold=0x05,SDThreshold=0x09,ARC=0x00,ARCInterval=0x00,OpticalSignalLevel=0xd1c8,LowOpThreshold=0xff,UppOpThreshold=0xff,OntRspTime=0x88b8,TranOpticLevel=0x034f,LowTranPowThreshold=0x81,UppTranPowThreshold=0x81)
```
Unfortunately I have no admin password to my ONT so cannot get PLOAM, i use default 1234567890

Anime4000, 2024-06-03:
```
omcicli mib get 84
omcicli mib get 171
```

got any output?

xMikooo, 2024-06-03:
> ```
> omcicli mib get 84
> omcicli mib get 171
> ```
> 
> got any output?

No, well yes the table header is displayed but no info on vlans

d5aqoep, 2025-01-04:
Several things can cause not registering.

Try with LOID set to 0 from Web GUI

For me setting Software version 1 & 2 allowed me to get VLAN
The telnet commands are below
cfgmib set OMCI_SW_VER1 3FE49362IJJK17
cfgmib set OMCI_SW_VER2 3FE49362IJJK17

Replace 3FE49362IJJK17 with your software version.

Other commands
flash set GPON_PLOAM_PASSWD 0123456789
flash set GPON_SN ALCLB4528576

flash set GPON_ONU_MODEL G-240G-E
flash set PON_VENDOR_ID ALCL
flash set HW_HWVER 3FE48299DDAC

belkone, 2025-03-05:
@xMikooo czy udało się coś zdziałać w temacie? :)