Anime4000/RTL960x #300: intel 82588es & HSGQ Xpon Stick

chiragkrishna, 2024-06-04:
Anyone using this combo? my intel card in unlocked to accept all modules but it is just not working with HSGQ stick. The lights also dont come up, tried forcing the stick to 1gbps but no luck. Is there anything which can be done?

Anime4000, 2024-06-04:
the PON Stick mostly use SGMII and HiSGMII in SerDes, the the Stick has own auto nego, have you wait around 5 to 10 minute ?

chiragkrishna, 2024-06-04:
that i didn't do, as soon as i plug the xpon stick that particular interface disappears from the list, so should i keep the stick in auto nego and wait for 10 mins??

chiragkrishna, 2024-06-05:
okay waited for 30mins, i dont think its doing anything, still no connection and no lights from that port!! tried with auto, 1gbps and 10g -> 1g.

chiragkrishna, 2025-03-20:
I went through it and my X520 is unlocked
[Guide](https://forums.servethehome.com/index.php?threads/patching-intel-x520-eeprom-to-unlock-all-sfp-transceivers.24634/)

@Anime4000 can you comment on this BASE-LX and BASE-PX?

z411, 2025-03-20:
@chiragkrishna Really? Is it working for you? I tried to patch my X520 and the offset 0x58 is already at FD (unlocked).
I also tried locking it (FC) and unlocking it (FD) again, and also using allow_unsupported_sfp in Linux and unsupported_sfp FreeBSD, but they all show "Unsupported SFP+ module" when loading the driver.

I also tried to set a specific speed in the stick but it never worked.

If it's working for you, what's your environment and what did you do to make it work?

chiragkrishna, 2025-03-20:
Doesn't work on x520.
Works on Nvidia connectx-4 takes a minute or 2.
May be @Anime4000 can help.

z411, 2025-03-20:
Oh my bad, I misunderstood your message.

Anime4000, 2025-03-20:
@chiragkrishna many have reported that Intel NIC especially modded in used market kind hard to get sync in 2.5Gbps Mode

ODI Stick V06 EEPROM can be edit, make sure checksum is valid before writing

perhaps follow OpenWRT patch which disable PX or LX and set 0x1F (31) in Nominal Bitrate, this will make Linux SFP fall into quirk patch to make 2.5G mode in SFP.

However, HiSGMII has no SFF Standard, only 2.5G Base-T and OC-48 (2500Mbps) has SFF Standard

z411, 2025-03-20:
@Anime4000 It doesn't work in 1G mode either.

> ODI Stick V06 EEPROM can be edit, make sure checksum is valid before writing

How can I know the version of my ODI stick? If so how do I write to the EEPROM?

chiragkrishna, 2025-03-20:
Network Adapter X520-DA2 only supports 10G/1G [Specs](https://www.intel.com/content/www/us/en/products/sku/39776/intel-ethernet-converged-network-adapter-x520da2/specifications.html)

z411, 2025-03-20:
> @z411 <https://github.com/z411> access the stick web portal, there you
> can find version

It shows V2.0 for me too. Isn't that too old?

chiragkrishna, 2025-03-20:
I don't know why there are so many versions, but will try to get the stick from fs.com

Untill then, there is some clue in [here](https://forums.servethehome.com/index.php?threads/patching-intel-x520-eeprom-to-unlock-all-sfp-transceivers.24634/page-6#post-452623)

z411, 2025-03-21:
I tried setting it to FF and same result. I also got a UFiber Instant module and it doesn't work either. I'm getting the ODI module and if that doesn't work I'll just replace the Intel card.

Anime4000, 2025-03-21:
@z411 try that,

what I know, almost all GPON stick I tried is using MII (media-independent interface)

SGMII = Serial Gigabit MII, 1Gbps
HiSGMII = High-speed Serial Gigabit MII, 2.5Gbps (¼ of XGMII)
XGMII = 10 Gigabit MII

this differ than NBase-T (2.5/5 RJ-45 Twisted Pair) and OC-48 (2488.32Mbps/2.5G)

Realtek do have somewhat 2500Base-X which similar to OC-48 but incomplete 

many Intel SFP+ NIC hard to get MII working, Broadcom NIC are better at this.

Last I read the SFF-8472, no mention about SGMII and HiSGMII identification, I guess Intel NIC standard follow this

z411, 2025-03-22:
I made it work!

[image]
[image]
[image]

Basically the hardware is compatible, it's just that the Intel driver is locked down and sucks. So I had to hack the ixgbe driver.

It will only recognize these SFP types: CU, LX, SX. Any other type it will reject the module. So I modded the driver to force it to accept it even if it doesn't recognize the SFP type, according to this article: https://medium.com/@pepijnvissers/patching-intels-sfp-driver-for-16a44f6daed1

But it still didn't work. And I noticed that the Intel card was trying to talk in 10GB to the stick, but obviously the stick is 1G. It seems that if it doesn't recognize the SFP type, it will also fall back to 10GB. So I edited the ixgbe_82599.c file so it will force it to 1G. And voila! Lights on, link up, ping works, webui works. I'll test some more and if everything works I can give detailed instructions.

The FreeBSD module is just a port of the Linux one so I will try to make the same edits and compile it there to use it on OPNsense.

chiragkrishna, 2025-03-22:
wow!! thats great!! you can send a PR to ixgbe driver if it works.

z411, 2025-03-22:
It works on FreeBSD & OPNsense too, the driver code is the same. I had to recompile kernel though, but it's easier on FreeBSD than Linux.

[image]
[image]

As you can see it says SFP type "Unknown" but it still works. I forced in the code to accept unknown modules as 1G SFP but the real fix would be to write code so it recognizes the real type of the ODI stick, but I'm not sure if Intel would want to upstream that if it's out-of-spec and non-Intel.

chiragkrishna, 2025-03-23:
definitely heading in right direction, if only there is a simpler method, instead of compiling the entire kernel. eagerly waiting for the steps 🤗

z411, 2025-03-24:
Another method posted by someone on Reddit was to overwrite the firmware of the ODI stick by using a Raspberry Pi by physically soldering it to the I2C pins of the module. At least compiling the kernel is simpler than that 😅 I will write instructions soon.

z411, 2025-03-26:
Investigated some more and it's because according to the SFF-8472 standard, byte 0x6 should indicate the SFP module Ethernet speed. In my module this byte has no bits set, thus the driver doesn't know what to do with it. What's strange though is that the dump here #286 has the 0x6 byte with the bit 1 set, meaning 1000BASE-LX. With this EEPROM, the ODI stick should work with the 82599ES out of the box. Not sure why the byte is empty in our sticks.

Anime4000, 2025-03-26:
@z411 this mainly for OpenWRT #250 
| Addr | Value | Note             |
|------|-------|------------------|
| 0x06 | 0x00  | Transceiver Type |
| 0x0a | 0x05  | Fiber Channel (100MB/s & 200MB/s) *not required*? |
| 0x0c | 0x1f  | Nominal Speed |

https://github.com/torvalds/linux/blob/cf1182944c7cc9f1c21a8a44e0d29abe12527412/drivers/net/phy/sfp-bus.c#L318-L336
```sfp-bus.c
	/* If we haven't discovered any modes that this module supports, try
	 * the bitrate to determine supported modes. Some BiDi modules (eg,
	 * 1310nm/1550nm) are not 1000BASE-BX compliant due to the differing
	 * wavelengths, so do not set any transceiver bits.
	 *
	 * Do the same for modules supporting 2500BASE-X. Note that some
	 * modules use 2500Mbaud rather than 3100 or 3200Mbaud for
	 * 2500BASE-X, so we allow some slack here.
	 */
	if (linkmode_empty(modes) && br_nom) {
		if (br_min <= 1300 && br_max >= 1200) {
			phylink_set(modes, 1000baseX_Full);
			__set_bit(PHY_INTERFACE_MODE_1000BASEX, interfaces);
		}
		if (br_min <= 3200 && br_max >= 2500) {
			phylink_set(modes, 2500baseX_Full);
			__set_bit(PHY_INTERFACE_MODE_2500BASEX, interfaces);
		}
	}
```

when `0x06` show `null` and `0x0c` in range `<= 3200` and `>= 2500`, Linux kernel will nego 2500 Base-X

The SFF-8472 need up to date to include `2500BASE-X` on `0x06`

for now, this hack works almost any Linux Router

z411, 2025-03-26:
Makes sense, thank you. Indeed my ODI stick has 0x0c set to 0x0d (13), meaning nominal 1300Mbps and should be seen as 1000BaseX by the driver. I will include both the hack and the proposed proper fix.

z411, 2025-03-26:
I wrote instructions to patch the driver here:

https://omaera.org/wlog/tech/odi_sfp

Anime4000, 2025-03-26:
> I wrote instructions to patch the driver here:
> 
> https://omaera.org/wlog/tech/odi_sfp

Very nice, I'll put into readme.md
btw, have you tried to make 2.5G work?

z411, 2025-03-26:
@Anime4000 From what I understand, this chip doesn't support 2.5G, only 1G and 10G:

[image]

Although I don't know if this is a hardware limitation, or a software limitation by the driver. If it's just the driver maybe it can be worked around, but I don't have any 2.5G capable device to change the ODI stick back to 1G in case it doesn't work.

Anime4000, 2025-03-26:
some broadcom also advertise as 1/10G, but hardware hack and NVRAM make it HiSGMII works.
hacking Intel Card going to be long journey

chiragkrishna, 2025-04-11:
did the drivers just got updated with this patch?? [ixgbe](https://lore.kernel.org/all/20240301184806.2634508-3-anthony.l.nguyen@intel.com/)
[opnsense](https://forum.opnsense.org/index.php?topic=46773.msg234614#msg234614)

z411, 2025-04-11:
The patch indeed adds support for reading the nominal rate when the stick advertises itself as BASE-BX10 (0x40 at addr 0x6), so it's a step forward for sure. But it still wouldn't make my stick work as it for some reason addr 0x6 has no bits set at all in the EEPROM of mine.

duerter, 2026-08-20:
> Doesn't work on x520. Works on Nvidia connectx-4 takes a minute or 2. May be [@Anime4000](https://github.com/Anime4000) can help.

Could you please provide the specific model of your CX4 NIC and the method used to get it working?
My hardware:
Computer: HP 800 G5 SFF
PCIe NIC A: MCX4121A-XCAT
PCIe NIC B: Intel X520-DA2 (82599ES)
PCIe NIC C: Dell (0)N20KJ (BCM57810S / QLogic QLE3442-CU-CK)
xPON ONU stick: ODI-DFP-34X-2C2 (HSGQ equivalent model/OEM)
Software/Firmware environment:
OS tested: Ubuntu 24, PVE 8/9, Windows 10
NIC firmware and drivers: Latest versions as of the time of this post
xPON ONU stick firmware: Tested with the latest and oldest versions for SFU, GHU, and Hybrid modes
xPON ONU stick settings: LAN_SDS_MODE = 1, 2, 3, 7
Specific symptoms:
After inserting the SFP module into the cage, NIC A—even when forced to 1G speed with auto-negotiation disabled—only detects the module's presence and retrieves module information (via `ethtool -m <interface>`).
NIC B loses the interface across all operating systems; a system reboot is required to restore it (however, if the xPON ONU stick remains in the cage, the interface does not reappear after the reboot).
NIC C works normally, provided the card has overcome initial cold-boot and motherboard PCH compatibility issues; however, after a subsequent power-off and cold boot, another reboot is required to resolve the compatibility issue before it functions correctly again.

@chiragkrishna

Gradius2, 2026-08-20:
It will work on Intel, but you need to do a mod, so it will accept any SFP on it

Gradius2, 2026-08-20:
Here:
https://forums.servethehome.com/index.php?threads/patching-intel-x520-eeprom-to-unlock-all-sfp-transceivers.24634/page-5

https://gist.github.com/ixs/dbaac42730dea9bd124f26cbd439c58e?permalink_comment_id=3418397

chiragkrishna, 2026-08-21:
@duerter you need an unlocked Intel 82599es and patched drivers. here is what i did for OPNSense

Patch Diff
```diff
diff --git a/sys/dev/ixgbe/ixgbe_phy.c b/sys/dev/ixgbe/ixgbe_phy.c
index 4b85b43d4c6..6596ce4f33b 100644
--- a/sys/dev/ixgbe/ixgbe_phy.c
+++ b/sys/dev/ixgbe/ixgbe_phy.c
@@ -1523,8 +1523,10 @@ s32 ixgbe_identify_sfp_module_generic(struct ixgbe_hw *hw)
                      hw->phy.sfp_type == ixgbe_sfp_type_1g_sx_core1 ||
                      hw->phy.sfp_type == ixgbe_sfp_type_1g_bx_core0 ||
                      hw->phy.sfp_type == ixgbe_sfp_type_1g_bx_core1)) {
-                       hw->phy.type = ixgbe_phy_sfp_unsupported;
-                       status = IXGBE_ERR_SFP_NOT_SUPPORTED;
+                       printf("IXGBE: custom XPON patch applied\n");
+                       hw->phy.type = ixgbe_phy_sfp_intel;
+                       hw->phy.sfp_type = ixgbe_sfp_type_1g_lx_core0;
+                       status = IXGBE_SUCCESS;
                        goto out;
                }
```

Auto Driver Patcher with notifications
```bash
#!/bin/sh
set -eu

GOTIFY_URL="https://gotify.url.com"
GOTIFY_TOKEN="GOTIFY_TOKEN"

PRIORITY=10
SMILEY="✅"
TITLE="$SMILEY OPNsense Xpon $SMILEY"
DELIMITER="--------------\n\n"

PATCH="/usr/local/bin/xpon.patch"

send_gotify_message() {
    message="$1"
    curl -s -S --data '{"message": "'"${message}"'", "title": "'"${TITLE}"'", "priority": '"${PRIORITY}"', "extras": {"client::display": {"contentType": "text/markdown"}}}' \
        -H 'Content-Type: application/json' \
        "$GOTIFY_URL/message?token=$GOTIFY_TOKEN" \
        >/dev/null 2>&1
}

SUCCESS=0

cleanup() {
    rc=$?

    if [ "$rc" -ne 0 ] && [ "$SUCCESS" -eq 0 ]; then
        send_gotify_message "$DELIMITER❌ **OPNSense:** XPON Driver Build FAILED"
    fi
}

trap cleanup EXIT

cd /usr/src

# Get current git state
OLD=$(git rev-parse HEAD 2>/dev/null || echo "none")

# Fetch/Update OPNsense source code
opnsense-code src

NEW=$(git rev-parse HEAD)

# Check if patch is already applied
PATCH_APPLIED=0
if git apply --reverse --check "$PATCH" >/dev/null 2>&1; then
    PATCH_APPLIED=1
fi

# If repository hasn't changed AND the patch is already there, we are done.
if [ "$OLD" = "$NEW" ] && [ "$PATCH_APPLIED" -eq 1 ]; then
    echo "No source changes and patch is already applied. Skipping."
    SUCCESS=1
    exit 0
fi

echo "Updating driver..."

# Clean repository state and apply patch
git reset --hard HEAD
git apply "$PATCH"

# Build the module
cd /usr/src/sys/modules/ix
make clean
make

# Ensure directory exists
mkdir -p /boot/modules

# Dynamically find the built module instead of hardcoding the full obj path
OBJ_PATH="$(make -V .OBJDIR)"
install -m 555 -o root -g wheel "$OBJ_PATH/if_ix.ko" /boot/modules/if_ix.ko

# Update kernel linker hints
kldxref /boot/modules

SUCCESS=1
send_gotify_message "$DELIMITER**OPNSense:** OPNsense XPON Driver Updated"
```

just create a Cron Job in OPNSense to run the script