Anime4000/RTL960x #303: ODI DFP-34X-2C2 very slow under load

AndreZiviani, 2024-06-15:
Hi
I've managed to get O5 and IP via DHCP (as expected) from my ISP but I have high ping when trying to download and specially upload

Original ONT: Technicolor TCG2232 (this is a custom model for my ISP AFAIK, original model is FGA2232CLB)
My plan: 500Mbps / 250Mbps
OLT: Huawei

I've tried a bunch of firmware versions and options, all with the same result...
ODI DFP-34X-2C2

Firmwares:
  - Anime4000: [M110_sfp_ODI_220923-240408_rel240515+mib.zip](https://github.com/Anime4000/RTL960x/issues/287#issuecomment-2111582306) (V1.1.8-240408 current), M114_sfp_ODI_hybrid_221209.tar, M114_sfp_ODI_hybrid_220916.tar, M110_sfp_ODI_220923.tar, M110_sfp_ODI_220817.tar
  - TripleOxigen: [custom V1.9.1-240202.tar](https://www.tripleoxygen.net/wiki/ont/odi/dfp-34x-2c2)
  - I believe I've tried some others from the issues

There are 5 vlans, the one used for internet is vlan `11` (the one I'm using) with the default `FwdOp` of `0x10`, I've tried `0x02` but got same results

`OMCC_VER`: `0x80` through `0xA3` works (with slow speed), `0xB0` through `0xB3` does not work, currently using `0x80`
`OMCI_OLT_MODE`: all of them works (with slow speed), currently using `1`
`OMCI_TM_OPT`: `0` and `2` works (with slow speed), `1` does not work, currently using `2`

I'am **not** trying to use the 2.5Gbps interface, it is connected to a Unifi Deam Machine Pro. I have another ODI stick with another ISP on the same UDMP and it is working fine.

I have obviously tested everything with the original ONT (Technicolor) and it works as expected, any help is **GREATLY** appreciated, please let me know if you need more info or want to test something

I have a good ping when the connection is idle (no download/upload):
```
# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=59 time=1.88 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=59 time=2.03 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=59 time=1.91 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=59 time=1.94 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=59 time=2.00 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=59 time=1.98 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=59 time=2.01 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=59 time=1.99 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=59 time=1.97 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=59 time=1.99 ms

--- 1.1.1.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9011ms
rtt min/avg/max/mdev = 1.883/1.971/2.033/0.043 ms
```

But when I start a download I don't get the full speed, only around half and ping increases 100x:
```
# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=59 time=1.93 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=59 time=1.96 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=59 time=1.90 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=59 time=2.00 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=59 time=2.30 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=59 time=43.1 ms     # <---  download starts
64 bytes from 1.1.1.1: icmp_seq=7 ttl=59 time=99.2 ms
64 bytes from 1.1.1.1: icmp_seq=8 ttl=59 time=147 ms
64 bytes from 1.1.1.1: icmp_seq=9 ttl=59 time=153 ms
64 bytes from 1.1.1.1: icmp_seq=10 ttl=59 time=135 ms
64 bytes from 1.1.1.1: icmp_seq=11 ttl=59 time=145 ms
64 bytes from 1.1.1.1: icmp_seq=12 ttl=59 time=151 ms
64 bytes from 1.1.1.1: icmp_seq=13 ttl=59 time=131 ms
64 bytes from 1.1.1.1: icmp_seq=14 ttl=59 time=148 ms
64 bytes from 1.1.1.1: icmp_seq=15 ttl=59 time=143 ms
64 bytes from 1.1.1.1: icmp_seq=16 ttl=59 time=144 ms     # <---  download stopped
64 bytes from 1.1.1.1: icmp_seq=17 ttl=59 time=1.90 ms
64 bytes from 1.1.1.1: icmp_seq=18 ttl=59 time=1.92 ms
64 bytes from 1.1.1.1: icmp_seq=19 ttl=59 time=1.86 ms
64 bytes from 1.1.1.1: icmp_seq=20 ttl=59 time=1.94 ms
^C
[... 275 lines cut ...]
      4 |     1562 |  ETH |     8
     64 |       26 | OMCI |    16
============================================================
```

</details>

<details>
  <summary>diag gpon show ds-flow</summary>

```
============================================================
     GPON ONU MAC D/S Flow Status
Flow ID | GEM Port | Type | Multicast | AES
      0 |     4095 |  ETH |         * |    
      1 |     1434 |  ETH |           |   *
      2 |     1690 |  ETH |           |   *
      3 |     1818 |  ETH |           |   *
      4 |     1946 |  ETH |           |   *
      5 |     1562 |  ETH |           |   *
     64 |       26 | OMCI |           |   *
============================================================
```

</details>

Anime4000, 2024-06-16:
try this to enable/add DSCP QoS:
```
omcicli mib set 171 0x01 DscpToPbitMapping $'\x01\x01\x01\x24\x92\x49\x49\x24\x92\x6d\xb6\xdb\x92\x49\x24\xb6\xdb\x6d\xdb\x6d\xb6\xff\xff\xff'
omcicli mib set 171 0x02 DscpToPbitMapping $'\x01\x01\x01\x24\x92\x49\x49\x24\x92\x6d\xb6\xdb\x92\x49\x24\xb6\xdb\x6d\xdb\x6d\xb6\xff\xff\xff'
```

AndreZiviani, 2024-06-16:
thanks but same result :(

after running both commands:
```
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x01
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4096, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 1, TPID 4
INDEX 1
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x010101
        0x249249
        0x492492
        0x6db6db
        0x924924
        0xb6db6d
        0xdb6db6
        0xffffff

Parsed DscpToPbitMapping:
dscp: 0 => pbit: 0
dscp: 1 => pbit: 0
dscp: 2 => pbit: 2
dscp: 3 => pbit: 0
dscp: 4 => pbit: 0
dscp: 5 => pbit: 4
dscp: 6 => pbit: 0
dscp: 7 => pbit: 1
dscp: 8 => pbit: 1
dscp: 9 => pbit: 1
dscp: 10 => pbit: 1
dscp: 11 => pbit: 1
dscp: 12 => pbit: 1
dscp: 13 => pbit: 1
dscp: 14 => pbit: 1
dscp: 15 => pbit: 1
dscp: 16 => pbit: 2
dscp: 17 => pbit: 2
dscp: 18 => pbit: 2
dscp: 19 => pbit: 2
dscp: 20 => pbit: 2
dscp: 21 => pbit: 2
dscp: 22 => pbit: 2
dscp: 23 => pbit: 2
dscp: 24 => pbit: 3
dscp: 25 => pbit: 3
dscp: 26 => pbit: 3
dscp: 27 => pbit: 3
dscp: 28 => pbit: 3
dscp: 29 => pbit: 3
dscp: 30 => pbit: 3
dscp: 31 => pbit: 3
dscp: 32 => pbit: 4
dscp: 33 => pbit: 4
dscp: 34 => pbit: 4
dscp: 35 => pbit: 4
dscp: 36 => pbit: 4
dscp: 37 => pbit: 4
dscp: 38 => pbit: 4
dscp: 39 => pbit: 4
dscp: 40 => pbit: 5
dscp: 41 => pbit: 5
dscp: 42 => pbit: 5
dscp: 43 => pbit: 5
dscp: 44 => pbit: 5
dscp: 45 => pbit: 5
dscp: 46 => pbit: 5
dscp: 47 => pbit: 5
dscp: 48 => pbit: 6
dscp: 49 => pbit: 6
dscp: 50 => pbit: 6
dscp: 51 => pbit: 6
dscp: 52 => pbit: 6
dscp: 53 => pbit: 6
dscp: 54 => pbit: 6
dscp: 55 => pbit: 6
dscp: 56 => pbit: 7
dscp: 57 => pbit: 7
dscp: 58 => pbit: 7
dscp: 59 => pbit: 7
dscp: 60 => pbit: 7
dscp: 61 => pbit: 7
dscp: 62 => pbit: 7
dscp: 63 => pbit: 7
=================================
=================================
EntityId: 0x02
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 14,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 0
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x010101
        0x249249
        0x492492
        0x6db6db
        0x924924
        0xb6db6d
        0xdb6db6
        0xffffff

Parsed DscpToPbitMapping:
dscp: 0 => pbit: 0
dscp: 1 => pbit: 0
dscp: 2 => pbit: 2
dscp: 3 => pbit: 0
dscp: 4 => pbit: 0
dscp: 5 => pbit: 4
dscp: 6 => pbit: 0
dscp: 7 => pbit: 1
dscp: 8 => pbit: 1
dscp: 9 => pbit: 1
dscp: 10 => pbit: 1
dscp: 11 => pbit: 1
dscp: 12 => pbit: 1
dscp: 13 => pbit: 1
dscp: 14 => pbit: 1
dscp: 15 => pbit: 1
dscp: 16 => pbit: 2
dscp: 17 => pbit: 2
dscp: 18 => pbit: 2
dscp: 19 => pbit: 2
dscp: 20 => pbit: 2
dscp: 21 => pbit: 2
dscp: 22 => pbit: 2
dscp: 23 => pbit: 2
dscp: 24 => pbit: 3
dscp: 25 => pbit: 3
dscp: 26 => pbit: 3
dscp: 27 => pbit: 3
dscp: 28 => pbit: 3
dscp: 29 => pbit: 3
dscp: 30 => pbit: 3
dscp: 31 => pbit: 3
dscp: 32 => pbit: 4
dscp: 33 => pbit: 4
dscp: 34 => pbit: 4
dscp: 35 => pbit: 4
dscp: 36 => pbit: 4
dscp: 37 => pbit: 4
dscp: 38 => pbit: 4
dscp: 39 => pbit: 4
dscp: 40 => pbit: 5
dscp: 41 => pbit: 5
dscp: 42 => pbit: 5
dscp: 43 => pbit: 5
dscp: 44 => pbit: 5
dscp: 45 => pbit: 5
dscp: 46 => pbit: 5
dscp: 47 => pbit: 5
dscp: 48 => pbit: 6
dscp: 49 => pbit: 6
dscp: 50 => pbit: 6
dscp: 51 => pbit: 6
dscp: 52 => pbit: 6
dscp: 53 => pbit: 6
dscp: 54 => pbit: 6
dscp: 55 => pbit: 6
dscp: 56 => pbit: 7
dscp: 57 => pbit: 7
dscp: 58 => pbit: 7
dscp: 59 => pbit: 7
dscp: 60 => pbit: 7
dscp: 61 => pbit: 7
dscp: 62 => pbit: 7
dscp: 63 => pbit: 7
=================================
```

AndreZiviani, 2024-06-17:
@Anime4000 just found something very wierd... this only happens when using IPv4, IPv6 works as expected (fast speed and low latency)

GuidoCHLM, 2024-06-27:
Hello guys. Need some help confirming if ODI DFP-34X-2C2 supports VEIP. I am from Argentina (Personal ISP) and need to replace a Huawei HG8245U ONT served by ALCA OLT. I tried a GPON-ONU-34-20BI but was not able to get it working. People here have successfully replaced the HG8245U with Ubifiber Loco so I am confident there should be a GPON stick that can work. Thanks!

Anime4000, 2024-06-28:
stick GPON-ONU-34-20BI, have you patched for Software Version to work?

GuidoCHLM, 2024-06-28:
Yes, patched SW version, still seems stick cannot figure out what to do with the VLANs. ME171 is empty. I attached the OMCI log in case you want to take a look. 
[omci.log](https://github.com/user-attachments/files/16023186/omci.log)

AndreZiviani, 2024-06-29:
hey @Anime4000 sorry for the delay, same thing happens on IPv6, any other ideas?

AndreZiviani, 2024-08-19:
hey @Anime4000 I've managed to get it working correctly with a HGU (`M114_sfp_ODI_hybrid_221209.tar`) instead of SFU, any ideas what it might be?