Anime4000/RTL960x #304: Is it possible to restore bricked V-SOL XPON V2801F

vbaskoff, 2024-07-03:
Hi Folks,

Without any reason sfp onu (V-SOL XPON V2801F) stopped to be reachable via IP, no mac address appeared behind sfp port in arp table as well.

RouterOS 'see' physical presence of the module and mark port as running.

Any way to bring it back to life?

`
[admin@MikroTik] > /interface/ethernet/monitor sfp5

# 1970-01-02 01:55:01 by RouterOS 7.12.1
# software id = 0MDF-UTYM
#
                      name: sfp5
                    status: link-ok
          auto-negotiation: done
                      rate: 1Gbps
               full-duplex: yes
           tx-flow-control: no
           rx-flow-control: no
                 supported: 10M-baseT-half,10M-baseT-full,100M-baseT-half,
                            100M-baseT-full,1G-baseT-half,1G-baseT-full,
                            1G-baseX
               advertising: 
  link-partner-advertising: 
        sfp-module-present: yes
               sfp-rx-loss: yes
                  sfp-type: SFP/SFP+/SFP28/SFP56
           eeprom-checksum: bad
                    eeprom: 0000: 03 00 00 00 00 00 00 00  03 00 00 00 00 00 00 00  ........ ........
                            *
                            0080: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
                            *
`
[image]

Anime4000, 2024-07-04:
only can see is via TTL UART

vbaskoff, 2024-07-04:
Does this sfp has some UART pins on the pcb?

Anime4000, 2024-07-04:
the SFP pin doesn't, need to open it and pin on the SoC pinout

crrodriguez, 2024-07-12:
I also have a "bricked" CarlitosPro stick which is the same device. Does not autoneg with any device I have.
 wrong LAN_SDS_MODE selected. Buying all the needed debug tools more expensive than new ODI stick.

dittohead, 2024-08-13:
@vbaskoff did you fix your dongle? got similar issue with fd511gx, but wait for needle pad

dittohead, 2024-08-16:
got same issue as @vbaskoff

```
/interface ethernet monitor sfp1  
                      name: sfp1
                    status: no-link
          auto-negotiation: done
               advertising: 
  link-partner-advertising: 
        sfp-module-present: yes
               sfp-rx-loss: no
           eeprom-checksum: bad
                    eeprom: 0000: 03 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
```

connect with uart sucessfuly, but can't do backup via uart
lan_sds_mode didn't help

@Anime4000 any ideas?

Anime4000, 2024-08-16:
did it boot successfully or stuck at recovery?

dittohead, 2024-08-16:
@Anime4000 successfull

Anime4000, 2024-08-16:
try enter this, see EEPROM is fixed:
```
diag i2c set port 1 dev 0x50 reg 0x00 data 0x03
diag i2c set port 1 dev 0x50 reg 0x01 data 0x04
diag i2c set port 1 dev 0x50 reg 0x02 data 0x01
diag i2c set port 1 dev 0x50 reg 0x03 data 0x00
diag i2c set port 1 dev 0x50 reg 0x04 data 0x00
diag i2c set port 1 dev 0x50 reg 0x05 data 0x00
diag i2c set port 1 dev 0x50 reg 0x06 data 0x02
diag i2c set port 1 dev 0x50 reg 0x07 data 0x22
diag i2c set port 1 dev 0x50 reg 0x08 data 0x00
diag i2c set port 1 dev 0x50 reg 0x09 data 0x01
diag i2c set port 1 dev 0x50 reg 0x0A data 0x00
diag i2c set port 1 dev 0x50 reg 0x0B data 0x01
diag i2c set port 1 dev 0x50 reg 0x0C data 0x0D
diag i2c set port 1 dev 0x50 reg 0x0D data 0x00
diag i2c set port 1 dev 0x50 reg 0x0E data 0x14
diag i2c set port 1 dev 0x50 reg 0x0F data 0xC8
diag i2c set port 1 dev 0x50 reg 0x10 data 0x00
diag i2c set port 1 dev 0x50 reg 0x11 data 0x00
diag i2c set port 1 dev 0x50 reg 0x12 data 0x00
diag i2c set port 1 dev 0x50 reg 0x13 data 0x00
diag i2c set port 1 dev 0x50 reg 0x14 data 0x41
diag i2c set port 1 dev 0x50 reg 0x15 data 0x6E
diag i2c set port 1 dev 0x50 reg 0x16 data 0x69
diag i2c set port 1 dev 0x50 reg 0x17 data 0x6D
diag i2c set port 1 dev 0x50 reg 0x18 data 0x65
diag i2c set port 1 dev 0x50 reg 0x19 data 0x34
diag i2c set port 1 dev 0x50 reg 0x1A data 0x30
diag i2c set port 1 dev 0x50 reg 0x1B data 0x30
diag i2c set port 1 dev 0x50 reg 0x1C data 0x30
diag i2c set port 1 dev 0x50 reg 0x1D data 0x2D
diag i2c set port 1 dev 0x50 reg 0x1E data 0x4E
diag i2c set port 1 dev 0x50 reg 0x1F data 0x69
diag i2c set port 1 dev 0x50 reg 0x20 data 0x6A
diag i2c set port 1 dev 0x50 reg 0x21 data 0x69
diag i2c set port 1 dev 0x50 reg 0x22 data 0x6B
diag i2c set port 1 dev 0x50 reg 0x23 data 0x61
diag i2c set port 1 dev 0x50 reg 0x24 data 0x00
diag i2c set port 1 dev 0x50 reg 0x25 data 0x00
diag i2c set port 1 dev 0x50 reg 0x26 data 0x00
diag i2c set port 1 dev 0x50 reg 0x27 data 0x00
diag i2c set port 1 dev 0x50 reg 0x28 data 0x55
diag i2c set port 1 dev 0x50 reg 0x29 data 0x4E
diag i2c set port 1 dev 0x50 reg 0x2A data 0x49
diag i2c set port 1 dev 0x50 reg 0x2B data 0x46
diag i2c set port 1 dev 0x50 reg 0x2C data 0x49
diag i2c set port 1 dev 0x50 reg 0x2D data 0x2D
diag i2c set port 1 dev 0x50 reg 0x2E data 0x4F
diag i2c set port 1 dev 0x50 reg 0x2F data 0x4E
diag i2c set port 1 dev 0x50 reg 0x30 data 0x55
diag i2c set port 1 dev 0x50 reg 0x31 data 0x2D
diag i2c set port 1 dev 0x50 reg 0x32 data 0x53
diag i2c set port 1 dev 0x50 reg 0x33 data 0x46
diag i2c set port 1 dev 0x50 reg 0x34 data 0x50
diag i2c set port 1 dev 0x50 reg 0x35 data 0x32
diag i2c set port 1 dev 0x50 reg 0x36 data 0x47
diag i2c set port 1 dev 0x50 reg 0x37 data 0x58
diag i2c set port 1 dev 0x50 reg 0x38 data 0x20
diag i2c set port 1 dev 0x50 reg 0x39 data 0x20
diag i2c set port 1 dev 0x50 reg 0x3A data 0x20
diag i2c set port 1 dev 0x50 reg 0x3B data 0x20
diag i2c set port 1 dev 0x50 reg 0x3C data 0x05
diag i2c set port 1 dev 0x50 reg 0x3D data 0x1E
diag i2c set port 1 dev 0x50 reg 0x3E data 0x00
diag i2c set port 1 dev 0x50 reg 0x3F data 0x6C
diag i2c set port 1 dev 0x50 reg 0x40 data 0x00
diag i2c set port 1 dev 0x50 reg 0x41 data 0x1A
diag i2c set port 1 dev 0x50 reg 0x42 data 0x00
diag i2c set port 1 dev 0x50 reg 0x43 data 0x00
[... 36 lines cut ...]
diag i2c set port 1 dev 0x50 reg 0x68 data 0x00
diag i2c set port 1 dev 0x50 reg 0x69 data 0x00
diag i2c set port 1 dev 0x50 reg 0x6A data 0x00
diag i2c set port 1 dev 0x50 reg 0x6B data 0x00
diag i2c set port 1 dev 0x50 reg 0x6C data 0x00
diag i2c set port 1 dev 0x50 reg 0x6D data 0x00
diag i2c set port 1 dev 0x50 reg 0x6E data 0x00
diag i2c set port 1 dev 0x50 reg 0x6F data 0x00
diag i2c set port 1 dev 0x50 reg 0x70 data 0x00
diag i2c set port 1 dev 0x50 reg 0x71 data 0x00
diag i2c set port 1 dev 0x50 reg 0x72 data 0x00
diag i2c set port 1 dev 0x50 reg 0x73 data 0x00
diag i2c set port 1 dev 0x50 reg 0x74 data 0x00
diag i2c set port 1 dev 0x50 reg 0x75 data 0x00
diag i2c set port 1 dev 0x50 reg 0x76 data 0x00
diag i2c set port 1 dev 0x50 reg 0x77 data 0x00
diag i2c set port 1 dev 0x50 reg 0x78 data 0x00
diag i2c set port 1 dev 0x50 reg 0x79 data 0x00
diag i2c set port 1 dev 0x50 reg 0x7A data 0x00
diag i2c set port 1 dev 0x50 reg 0x7B data 0x00
diag i2c set port 1 dev 0x50 reg 0x7C data 0x00
diag i2c set port 1 dev 0x50 reg 0x7D data 0x00
diag i2c set port 1 dev 0x50 reg 0x7E data 0x00
diag i2c set port 1 dev 0x50 reg 0x7F data 0x00
```

crrodriguez, 2024-08-16:
Also the firmware seems to support TFTP flash/boot but I couldn't make it work. there was no TFTP client traffic from the device een when all addresses are correct. I guess this is because the device actually boots but does not work as opposed to be bricked and the fallback does not trigger.

dittohead, 2024-08-16:
@Anime4000 not looks like it helps. same eeprom in mikrotik monitoring.

>

dittohead, 2024-08-18:
UPD: calculate and set VS_AUTH_KEY + switch LAN_SDS_MODE from 1 to 4 - didn't help
check eeprom from `diag i2c get ...` - all is stored, but mikrotik interface still displays as 
`0000: 03 00 00 00 00 00 00 00`

dittohead, 2024-08-20:
btw @Anime4000, this happened after [this](https://github.com/Anime4000/RTL960x/issues/312) issue