Anime4000/RTL960x #305: [Feature Request] FTP server for easy file transfers for ODI DFP-34X-2C2

bademux, 2024-07-05:
Hi,
In some cases netcat to host isn't the solution, eq. NAT'ed stick.
Will it be possible to share precompiled ftp server, I think, that something like tftpd is good enough.
It will be great to embed (manual launch) on customized firmware.

Thanks.

Anime4000, 2024-07-06:
that would be nice, do you know how to compile for MIPS R3000 ?

bademux, 2024-07-06:
@Anime4000 I assume there some toolchain to be setup. It will be great to have instructions for it.

bademux, 2024-07-06:
Well, it will be interesting to simplify customisation. If I'm remember it right, I saw qemu for the stick.

Anime4000, 2024-07-06:
yes, I wrote that script for unpack and repacking, just that

lehoangnb, 2024-08-09:
Hi, I've compiled a lightweight ftp binary for mips, just prebuilt binary to /bin of firmware, chmod 755 and execute it or using nc to push it to /tmp
[image]
[ftp.zip](https://github.com/user-attachments/files/16559120/ftp.zip)