Anime4000/RTL960x #310: Please update info that DFP-34X-2C2 and DFP-34X-2C3 is the same device but with different optics/connector (UPC vs APC)

damex, 2024-08-09:
whole difference between them out of box is basically 'name' and different optics.
vendor from china confirmed that that DFP-34X-2C2 use SC/UPC and DFP-34X-2C3 use SC/APC and that is all there is to it. both run Realtek nowaday, have supposed to have same firmware and internals beside optics ;)

i have purchased pair of DFP-34X-2C3 from local ecommerce platform (technically shipped from taobao) and received them.
qr sticker have different placement and some cosmetic sticker difference but firmware seem to be the same.

i reflashed them with [M110_sfp_ODI_220923.tar](https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923.tar) and they have been working well with VNPT here in Vietnam.
original firmware complained about firmware flash attempt (something about missmatch) at first but after doing factory reset it accepted customized firmware (actually it was originally running 'the same' 220923 firmware). i did one more reset after flashing with firmware above and configured for my use. everything been working well since.

ianmuninio, 2024-08-11:
I can confirm this. I'm using the DFP-34X-2C3 which uses SC/APC the vendor mentioned that DFP-34X-2C2 uses SC/UPC connector.

cuongdt1994, 2024-08-12:
> whole difference between them out of box is basically 'name' and different optics. vendor from china confirmed that that DFP-34X-2C2 use SC/UPC and DFP-34X-2C3 use SC/APC and that is all there is to [...]

I also use VNPT but still can't connect.

OLT: ALCL (Alcatel-Lucent)

As far as I know, you only need to enter SN and LOID to dial pppoe, right?

If wrong, how do you dial pppoe with stick, can you guide me?

Anime4000, 2024-08-12:
Converting this as discussion