Anime4000/RTL960x #314: MAP-E tunnel not working with VSOL V2802RH or HSGQ-Z110 on Japan NURO Hikari 2Gbps plan

missing233, 2024-09-01:
I'm having some trouble with my NURO Hikari 2Gbps connection which uses an IPv4 over IPv6 tunnel based on MAP-E protocol. 
When I'm using VSOL V2802RH or HSGQ-Z110(I've already set nas0_0 to bridged), I can only get the DHCPv6 for IPv6 to set up properly. 
(I followed this script: https://github.com/site-u2023/config-software/blob/main/map-e-nuro.sh)

MAP-E tunnel just doesn't seem to work, it's not receiving any packets. When I switch to GPON Stick like Huawei MA5671A everything works fine.

Any idea what's going on here or how to fix it?

Anime4000, 2024-09-02:
that script for OpenWRT based,

I bet VSOL V2802RH only do bridge the VLAN, perhaps check MIB 84 and 171 ?

missing233, 2024-09-03:
> that script for OpenWRT based,
> 
> 
> 
> I bet VSOL V2802RH only do bridge the VLAN, perhaps check MIB 84 and 171 ?

There’s no issue with IPv6; the PD can be obtained just fine. MIB seems to be working fine too. not sure why the 4in6 tunnel is acting up.