Anime4000/RTL960x #315: Step-by-Step update firmware only access UART

d0bro, 2024-09-18:
Hello, i dont have accsess into webGui on my **ODI Realtek DFP-34G-2C2** 

I tried to switch the slot (**setenv**) and **reset_sw**. The commands in tty console are different from telnet. 
I run stty and screen on my MacOS:
```bash
stty -f /dev/tty.usbserial-130 -cstopb -parity -ixon raw speed 115200
screen /dev/tty.usbserial-130 115200
```

**Is there a detailed instruction on how to flash it to your firmware?**

I understand that it is possible to do this somehow via tftp.
Is it possible to see what IP the webgui is configured on via UART console?

It is my printenv:

```
b0=setenv bootargs ${bootargs_base} ${mtdparts0} ${rst2dfl_flg}; bootm ${img0_kernel}
b1=setenv bootargs ${bootargs_base} ${mtdparts1} ${rst2dfl_flg}; bootm ${img1_kernel}
baudrate=115200
boot_by_commit=if itest.s ${sw_commit} == 0;then run set_act0;run b0;else run set_act1;run b1;fi
boot_by_tryactive=if itest.s ${sw_tryactive} == 0;then setenv sw_tryactive 2;setenv sw_active 0;saveenv;run en_wdt;run b0;else setenv sw_tryactive 2;setenv sw_active 1;saveenv;run en_wdt;run b1;fi
bootargs_base=console=ttyS0,115200
bootcmd=if itest.s ${sw_tryactive} == 2; then run boot_by_commit;else run boot_by_tryactive;fi
bootdelay=5
en_wdt=mw b8003268 e7c00000
erase_cfgfs=sf erase ${fl_cfgfs} +${fl_cfgfs_sz}
erase_env=sf erase ${fl_env} +${fl_env_sz};sf erase ${fl_env2} +${fl_env_sz}
ethact=LUNA GMAC
ethaddr=00:E0:4C:86:70:01
fileaddr=80000000
filesize=39A800
fl_boot_sz=40000
fl_cfgfs=44000
fl_cfgfs_sz=3c000
fl_env=40000
fl_env2=42000
fl_env_sz=2000
fl_kernel1=80000
fl_kernel1_sz=14c000
fl_kernel2=440000
fl_kernel2_sz=14c000
fl_rootfs1=1cc000
fl_rootfs1_sz=274000
fl_rootfs2=58c000
fl_rootfs2_sz=274000
fx1000_init=mw bb000084 00000048
img0_kernel=94080000
img1_kernel=94440000
ipaddr=192.168.1.3
mtdparts0=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),1328K(k0)ro,2512K(r0)ro,1328K(k1),2512K(r1),4K@0ro,4K@0ro,4K@0ro,4K@0ro,1328K@512K(linux),2512K@1840K(rootfs) root=31:5
mtdparts1=mtdparts=rtk_spi_nor_mtd:256K(boot),8K(env),8K(env2),240K(config),1328K(k0),2512K(r0),1328K(k1)ro,2512K(r1)ro,4K@0ro,4K@0ro,4K@0ro,4K@0ro,1328K@4352K(linux),2512K@5680K(rootfs) root=31:7
mupgrade_en=1
netmask=255.255.255.0
serverip=192.168.1.101
set_act0=if itest.s ${sw_active} != 0;then setenv sw_active 0;saveenv;fi
set_act1=if itest.s ${sw_active} != 1;then setenv sw_active 1;saveenv;fi
sgmii_init=mw bb000084 00000044
stderr=serial
stdin=serial
stdout=serial
sw_active=0
sw_commit=0
sw_crc0=5bfcffa5
sw_crc1=5bfcffa5
sw_tryactive=2
sw_valid0=1
sw_valid1=1
sw_version0=V1.0-220923
sw_version1=V1.0-220304
tftp_base=80000000
upb=tftp ${tftp_base} plr.img && crc32 ${fileaddr} ${filesize} && sf erase 0 +${fl_boot_sz} && sf write ${fileaddr} 0 ${filesize}
upe=tftp ${tftp_base} uboot-env-98d-eng.bin && sf erase ${fl_env} +${fl_env_sz} && sf write ${fileaddr} ${fl_env} ${fl_env_sz} && sf erase ${fl_env2} +${fl_env_sz} && sf write ${fileaddr} ${fl_env2} ${fl_env_sz}
upk=tftp ${tftp_base} uImage && crc32 ${fileaddr} ${filesize} && sf erase ${fl_kernel1} +${fl_kernel1_sz} && sf write ${fileaddr} ${fl_kernel1} ${filesize}
upk1=tftp ${tftp_base} uImage && crc32 ${fileaddr} ${filesize} && sf erase ${fl_kernel2} +${fl_kernel2_sz} && sf write ${fileaddr} ${fl_kernel2} ${filesize}
upr=tftp ${tftp_base} rootfs && crc32 ${fileaddr} ${filesize} && sf erase ${fl_rootfs1} +${fl_rootfs1_sz} && sf write ${fileaddr} ${fl_rootfs1} ${filesize}
upr1=tftp ${tftp_base} rootfs && crc32 ${fileaddr} ${filesize} && sf erase ${fl_rootfs2} +${fl_rootfs2_sz} && sf write ${fileaddr} ${fl_rootfs2} ${filesize}
upt=tftp 80000000 img.tar && upimgtar ${fileaddr} ${filesize}
upv=tftp 80000000 vm.img;upvmimg ${fileaddr}
yk=loady 80000000 && cp.b 80000000 81000000 ${filesize} && cmp.b 80000000 81000000 ${filesize} && sf erase ${fl_kernel1} +${fl_kernel1_sz} && sf write 80000000 ${fl_kernel1} ${filesize}
yr=loady 80000000 && cp.b 80000000 81000000 ${filesize} && cmp.b 80000000 81000000 ${filesize} && sf erase ${fl_rootfs1} +${fl_rootfs1_sz} && sf write 80000000 ${fl_rootfs1} ${filesize}
yu=loady 80000000 && cp.b 80000000 81000000 ${filesize} && cmp.b 80000000 81000000 ${filesize} && sf erase 0 +${fl_boot_sz} && sf write 80000000 0 ${filesize}

Environment size: 3769/8187 bytes
```

**ethaddr** - differs from what is indicated on the sticker itself

Anime4000, 2024-09-19:
have you tried to boot different partition?

based on current env, `sw_active=0`, `sw_commit=0` try switching to `1`

most of the time it work by fall back to correct working firmware partition

d0bro, 2024-09-19:
Yes, I tried it. No result(

Anime4000, 2024-09-19:
can share the boot log?