Anime4000/RTL960x #317: HSGQ-XPON-Stick ONU SPI Flash file

Sonic-81, 2024-09-23:
Can someone please provide me with the HSGQ-XPON-Stick APC SPI Factory Flash file, my SFP had got bricked so I fixed it with the ODI modified SPI bin file provided in #286 but I want the unmodified factory SPI bin file.

Sonic-81, 2024-09-24:
Hey Guys, I had taken the backup of MTD7 file which contains the original firmware which came with the SFP stick i.e. V1.0--230303 is there any way I can convert it to .bin  or .tar or any other way to restore it.

Anime4000, 2024-09-24:
you can copy mtd7 to mtd5 I believe,

if you want to take mtd7 as firmware file, you need a Linux, prefer Ubuntu

take mtd6 (uImage) and mtd7 (rootfs)
with name `uImage` and `rootfs`, pack it as tar file
then use my `qemu-test.sh`, you might need take `fwu_ver` and `fwu.sh` from working firmware

Sonic-81, 2024-09-25:
Hi, I am new to Linux, I booted with Ubuntu live usb and tried the above steps but when I am trying to run the qemu-test.sh file as qemu-test.sh <packagename> I am getting error qemu-test.sh: command not found.
Attached is the tar file I created. I tried installing the firmware but it gave error File size exceeded.
Please help.
[230303.zip](https://github.com/user-attachments/files/17128935/230303.zip)