Anime4000/RTL960x #318: DLINK DPN-100 Ver:C1

BOPOH1, 2024-09-28:
Привет! Может кто-нибудь мне расшифровать какие VLAN приходят ко мне на SFP?

BOPOH1, 2024-09-28:
#ONT/system/mib>show 84
Table VlanTagFilterData, VLAN Tag Filter Data, total 6 instances

EntityID                  = 0x0002
FilterTbl                 = 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 1

EntityID                  = 0x0003
FilterTbl                 = 00 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 1

EntityID                  = 0x0004
FilterTbl                 = 00 1e 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 2

EntityID                  = 0x0005
FilterTbl                 = 00 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 1

EntityID                  = 0x0006
FilterTbl                 = 00 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 1

EntityID                  = 0x0007
FilterTbl                 = 00 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FwdOp                     = 16
NumOfEntries              = 1

Anime4000, 2024-09-29:
#ONT/system/mib>show 171
then paste here
https://hack-gpon.org/gpon-omci-vlan-parser/

BOPOH1, 2024-09-29:
Filter outer priority	14
Filter outer VID	0
Filter outer TPID	NaN
Filter inner priority	0
Filter inner VID	0
Filter inner TPID	1
Filter ether type	0
Treatment tags to remove	3
Treatment outer priority	0
Treatment outer VID	20
Treatment outer TPID	NaN
Treatment inner priority	15
Treatment inner VID	0
Treatment inner TPID	3
----------------------------------------
Filter outer priority	0
Filter outer VID	0
Filter outer TPID	NaN
Filter inner priority	0
Filter inner VID	0
Filter inner TPID	5
Filter ether type	1
Treatment tags to remove	0
Treatment outer priority	0
Treatment outer VID	0
Treatment outer TPID	NaN
Treatment inner priority	13
Treatment inner VID	0
Treatment inner TPID	0
----------------------------------------
Filter outer priority	0
Filter outer VID	26
Filter outer TPID	NaN
Filter inner priority	0
Filter inner VID	0
Filter inner TPID	NaN
Filter ether type	0
Treatment tags to remove	0
Treatment outer priority	0
Treatment outer VID	0
Treatment outer TPID	NaN
Treatment inner priority	0
Treatment inner VID	32
Treatment inner TPID	1
------------------------------------------
Filter outer priority	13
Filter outer VID	0
Filter outer TPID	NaN
Filter inner priority	0
Filter inner VID	22
Filter inner TPID	NaN
Filter ether type	0
Treatment tags to remove	0
Treatment outer priority	0
Treatment outer VID	0
Treatment outer TPID	0
Treatment inner priority	0
Treatment inner VID	0
Treatment inner TPID	0
-------------------------------------------
Filter outer priority	0
Filter outer VID	0
Filter outer TPID	0
Filter inner priority	0
Filter inner VID	0
Filter inner TPID	0
Filter ether type	0
Treatment tags to remove	0
Treatment outer priority	0
Treatment outer VID	0
Treatment outer TPID	0
Treatment inner priority	0
Treatment inner VID	0
Treatment inner TPID	NaN

BOPOH1, 2024-09-29:
Filter outer priority	14
Filter outer VID	0
Filter outer TPID	NaN
Filter inner priority	0
Filter inner VID	0
Filter inner TPID	1
Filter ether type	0
Treatment tags to remove	3
Treatment outer priority	0
Treatment outer VID	20
Treatment outer TPID	NaN
Treatment inner priority	15
Treatment inner VID	0
Treatment inner TPID	3

BOPOH1, 2024-09-29:
So I did as you said. Can you help decipher it for me?